raspberrypi/rpi-imager #400: [OS Request] HotSpotOS

guysoft, 2022-03-20:
A Raspberry Pi](http://www.raspberrypi.org/ distribution that makes a Raspberrypi start a hotspot, if no wifi was found to conenct to. It also bridges internet comming in from ethernet.

The hotspot scripts are based on the [scripts by roboberry](http://www.raspberryconnect.com/network/item/330-raspberry-pi-auto-wifi-hotspot-switch-internet) . And have been improved to be more runbust and support ipv6 with prefix delegation as well as the existing ipv4. They are part of the auto-hotspot module of [CustomPiOS](https://github.com/guysoft/CustomPiOS). HotSpotOS is based on CustomPiOS.

https://github.com/guysoft/HotSpotOS

```json
    "os_list": [
        {
            "name": "HotSpotOS",
            "description": "Start a hotspot, bridges internet comming in from ethernet.",
            "icon": "https://raw.githubusercontent.com/guysoft/HotSpotOS/devel/media/rpi-imager-HotSpotOS.png",
            "website": "https://github.com/guysoft/HotSpotOS",
            "subitems_url": "https://unofficialpi.org/rpi-imager/rpi-imager-hotspotos.json"
        }
    ]
```

tdewey-rpi, 2023-11-17:
Please fill out the form to request an addition to the OS list at https://forms.gle/cbsU27aLLVaxUXbw8.

closed 2023-11-17 by tdewey-rpi

guysoft, 2023-11-17:
@tdewey-rpi Hey there are quite a few OSes I submitted.
Is there a point to re-submit them in the form, for example, some criteria they should meet?

I just rather not put all the work in if its just going to get stuck.

Also, you might like to see what I did on pi-imager. Which is the unofficial fork that serves these images and others who were submitted to it. It uses CI-CD to update all the snippets. So people can send you a PR that once you merge updates all the stuff automatically:
https://github.com/guysoft/pi-imager-web/

Which then gets updated to Pi imager, which in turn has CI/CD for mac/windows/linux + Appimage for linux:
https://github.com/guysoft/pi-imager/

List of OSes submitted and maintain just so you get the idea:

* https://github.com/raspberrypi/rpi-imager/issues/392
* https://github.com/raspberrypi/rpi-imager/issues/382
* https://github.com/raspberrypi/rpi-imager/issues/389
* https://github.com/raspberrypi/rpi-imager/issues/388
* https://github.com/raspberrypi/rpi-imager/issues/387
* And the one here of course: https://github.com/raspberrypi/rpi-imager/issues/400

tdewey-rpi, 2023-11-17:
Yes, the form explicitly asks a number of security related questions and collects named points of contact for each OS submitted.

This is a change from how we've previously operated, but one that is required to ensure we can roll out new Imager features in a coherent manner.

tdewey-rpi, 2023-11-17:
Regarding the automatic listing, I am not able to commit to such a mechanism in this project.

As the Raspberry Pi Imager, this project has our brand associated with it - and we have to be somewhat careful about what OS we guide our customers into using. A side effect of this change of flow is that we now ask about the security of the OS you're providing, and expect to get a reasoned answer for your security posture. This is all aimed at offering a better guidance for new users - a trend that I've been unambiguous in pursuing.

I hope that helps explain my positioning. I'm happy to entertain a reasoned counterpoint, and rarely consider my views fixed, but I have already put some considerable thought into the risks in play.

guysoft, 2023-11-18:
Ah of course. What I meant was - You can build upon it your own flow.

The pipeline code there lets you manage the PRs, and once they are approved all the work of adding them is done for you.
Its automation, not automatic :)