raspberrypi/rpi-imager #403: [OS Request] ElectricSheepPi

guysoft, 2022-03-21:
A distribution to run [Electric Sheep : Crowdsourced Evolving Art](https://electricsheep.org/) out of the box and the scripts necessary to load it at boot. 

Image build sources: https://github.com/guysoft/ElectricSheepPi

```json
    "os_list": [
        {
            "name": "ElectricSheepPi",
            "description": "A Raspberry Pi distribution to to run Electric Sheep, crowdsourced evolving art",
            "icon": "https://raw.githubusercontent.com/guysoft/ElectricSheepPi/devel/media/rpi-imager-electricsheep.png",
            "website": "https://electricsheep.org",
            "subitems_url": "https://unofficialpi.org/rpi-imager/rpi-imager-electricsheeppi.json"
        }
    ]
```

ghollingworth, 2022-03-22:
This should just be apt installed...  It's not valid to add a whole image because someone hasn't spent the time to package it.

closed 2022-03-22 by ghollingworth

guysoft, 2022-03-22:
It requires:
1. Compiling from source
2. It uses CustomPiOS gui module that configures the gui application to start on boot of the ui.

Ill what is included in other images too