raspberrypi/rpi-imager #404: No Documentation on how to set up SSH Key in imager

RFBomb, 2022-03-23:
I tried setting up a new image of Octoprint tonight, only to fail multiple times because once it was set up, I had no way to access the headless device. 
I have previously successully used SSH keys on this device before, but figured since I was flashing a new SD I would try out the built-in setup to do that for me. Too bad there is zero documentation on what you are actually supposed to do here. 
The field came pre-filled with a single-line key. 

- Is that supposed to be changed to a custom generated key?
   - Since this key was generated seemingly by the installer, how do I use it? Do I past it into a PPK file? Do I paste it into a "public_key" file somewhere on the computer? Theres nothing telling you how it came up with this string, or how to use it successfully. 
   - I assumed it pulled it from my computer and it would just work. If this wasn't the case, why even autofill the field?! 
      - ( Obviously it did not work. Instead I got `Permission denied (publickey)` whenever attempting to log in.
      
- Can I use a key that was previously generated using PuttyGen? 
   - If so, how? Just paste the public key in? 
   - If this is the case, this needs to be detailed somewhere. I couldn't find any information on this, and the installer itself doesn't seem to have any documtenation at all about the advanced settings. None that I was able to locate on the internet anyway. Most are straightforward, but the SSH key is a bit more advanced than the rest, and as such should bear minimum have a `?` button nearby it. Or an easily locatable segment on the website (such as here: https://www.raspberrypi.com/documentation/computers/remote-access.html#introduction-to-remote-access )

lurch, 2022-03-23:
ping also @aallan

aallan, 2022-03-23:
Honestly not something I've done before, I'll ask around and see how it works. Ping @ghollingworth to transfer this issue to the documentation repo.

RFBomb, 2022-03-23:
I saw several other issues that mention this ssh-key and their troubles. 

So while having it on the website document is good, I still think it needs some sort of clarification in the UI itself. Even if that is a simple link to the documentation page. 

I think this is the only real option on that page right now that isn't straightforward on how to use it. (Well, overscan too, but that atleast has a note in the Raspi-config docs). 

From a UI standardization perspective, they'd all have a tooltip or a button to click for more info, but I think that's unnecessary for things like the username, password, WiFi, etc that are all pretty self-explanatory and would just introduce clutter. 

Anyways, looking forward to seeing a write up on how it (ssh key option) actually works and what the end-user needs to do to use it successfully. If it works well, this would be a big time saver over older methods of manually adding the keys in.