raspberrypi/rpi-imager #405: Raspberry pi imager return 404 while flashing OS in SD Card

Debagnik, 2022-03-25:
I have a Raspberry Pi 400 4GB with Ubuntu Server 21.10 64-bit OS installed.
I decided to install ubuntu server again because I broke the OS so I decided to reinstall the OS using Pi Imager utility tool,
but for some reason I can't download the OS from the servers using the tool attached a Screenshot.

Error says the server returns 404 code. but when I use the same IP to browse using chrome I can see the contents of the file.

although I can successfully install raspberry pi OS using an .img file

please help

Version: 
raspberry Pi imager 1.5.1
trying to flash Ubuntu Server 21.10 - 64-bit
as Windows 11- 64-bit Host computer
on a 16 GB microSD card

[![Image Attachment with chrome showing the content of the IP address][1]][1]

  [1]: https://i.stack.imgur.com/tCfvR.png

maxnet, 2022-03-25:
>The error says the server returns a 404 code. but when I use the same IP to browse using chrome I can see the contents of the file.

Please open a terminal, and start "rpi-imager" from there.
It should output the full URL (not just the IP) it tries to download and server response to console.

Pukimaa, 2022-08-11:
> > The error says the server returns a 404 code. but when I use the same IP to browse using chrome I can see the contents of the file.
> 
> Please open a terminal, and start "rpi-imager" from there. It should output the full URL (not just the IP) it tries to download and server response to console.

It's `Image URL: "http://cdimage.ubuntu.com/releases/jammy/release/ubuntu-22.04-preinstalled-server-arm64+raspi.img.xz"`

maxnet, 2022-08-11:
>It's Image URL: "http://cdimage.ubuntu.com/releases/jammy/release/ubuntu-22.04-preinstalled-server-arm64+raspi.img.xz"

Seems the 22.04 image no longer exist on Ubuntu's servers, because it got replaced by 22.04.1
https://cdimage.ubuntu.com/releases/jammy/release/ubuntu-22.04.1-preinstalled-server-arm64+raspi.img.xz does exist.

@waveform80
Ubuntu should update their json: https://changelogs.ubuntu.com/raspi/os_list_imagingutility_ubuntu.json

waveform80, 2022-08-11:
@maxnet Yup, one of my colleagues is on it (I'm on holiday this week -- had intended to be around for the release last week, but it got pushed due to some last minute critical issues). Should be fixed later today (he says, keeping a vague eye on IRC ;)

EML-github, 2022-08-11:
Curiously, I've just had exactly this problem. Any chance that your colleague could manage this before I finish my coffee? :)

EML-github, 2022-08-11:
Working now, thanks.

fgarcia-i40, 2022-09-01:
Hello,

I have the very same issue with Ubuntu Server 20.04.4 LTS. Looks like the image is already for 20.04.5 but Ubuntu json hasn't been updated yet.

Can you please check into it?

Thanks!

waveform80, 2022-09-01:
@fgarcia-i40 yup, the 20.04.5 point release just got released (less than an hour ago); I've sent the merge request to update our JSON but there's always a bit of a delay between the images appearing and the JSON getting merged. Should be fixed shortly

wkoeter, 2022-09-01:
Its available now

Pukimaa, 2022-09-06:
Was resolved within 1 or 2 days actually. What about closing this issue now?

closed 2022-09-06 by Debagnik