raspberrypi/rpi-imager #406: rpi-imager keeps changing password in settings

mschipperheyn, 2022-03-26:
I don't know how many times I've burned an image and get a password denied. Opening settings, shows different sizes of the password field. Which cannot be disabled and which doesn't permit seeing the password. My guess is, when you change it here, it adds 'raspberry' to whatever you put there in some cases. Please:
- allow disabling password field: guarantee the default raspberry
- implement a show password option

maxnet, 2022-03-26:
>I don't know how many times I've burned an image and get a password denied.

What operating system image is "an image", and did you also try logging in locally, or only remotely?
If remotely, you did enter right username "ssh username@hostname.local" ?

>Opening settings, shows different sizes of the password field.

That is expected.
Passwords are stored hashed.
We cannot recover the original.

qkum, 2022-04-05:
It works fine.

There is even a button to "show" the password in settings, as he claims there is not - I used it 60 seconds ago in the current newest version 1.7.2 and it works.

Sounds like he calls "program files" for "settings" as @maxnet spotted.

Different password length when hashed must be so an attacker don't know the password length. Basic IT Security.
- Knowing the password length makes it 10000 times easier to bruteforce with software.

* On Linux I have experienced password bugs from using special characters !&#(/# and special non standard letters And empty spaces in the password. Maybe that is what is his real problem and he just blames someone else thinking he can't be the root cause of the error? (sounds quite like it)

qkum, 2022-04-05:
Ok, I know what he may be talking about now.

1) Yes, you can "show password" with the WIFI password.
2) But you can't click "show password" with the Password for Root/Admin.
3) He talked about the Wifi password. And him not having any "show password" option. But 1.7.2 on my screen, it is there.
4) But he is on to something though. By random chance I clicked the "show password" for the Wifi. And the App had as he said changed the wifi password by itself. To something like "eafawfawgawgjuinhpdro" gibberish.
5) I just updated from 1.7.1 to 1.7.2 before writing in this thread. So, maybe it was because of updating the app? 
6) I updated the app by downloading the new one from this github repo and installing it on top of the old one and it seemed to work. 
7) But the SD card I flashed with the newest 32 bit Raspberry Pi OS actually had impossible to solve wifi problems so I just flashed a new SD card with Buster instead and notice what I just described.
8) What kind of wifi problems? - It would connect to my wifi, connect VNC to control my Pi from my pc, and then lose connection within seconds, and 30 sek after lose the internet connection. Rebooting changed nothing. Unplugging everything changed nothing. apt upgrade changed nothing. Change wifi with sudo raspi-config also gave the same result. My Wifi is within of 1 meter of my Raspberry Pi 4b. There are no wifi problems whatso ever in reality and my pc proves it. It kept losing the wifi and using 10 min to reconnect for no reason.

I hope my experience / feedback can be helpful.

qkum, 2022-04-05:
Ok, the 2d Flashed SD Card / Image with the newest 32 bit Raspberry Pi OS also epic failed on keeping a stable Wifi connection.

- I then deleted the "Raspberry Pi Imager", Reinstalled "Raspberry Pi Imager".
- Flashed a 3d SD Card again with the newest 32 bit Raspberry Pi OS. 
- Tested this SD Card on my Raspberry Pi 400 first, and it worked like a charm.
- Then I tested it on the first Raspberry Pi 4b, with all the Wifi problems,  (that I got from Amazon today, so I suspected a defective Wifi Chip).
- Now it freaking works like a charm! 

`* It was the freaking Raspberry Pi Imager 1.7.1 upgraded to 1.7.2 without uninstalling the old version first that was the problem! `

* It must have written something to my SD Cards that bugged the Wifi out like crazy!  (on 2 Flashed SD Cards in a row!)
* You would think that it is impossible but it is clearly not! 
* Option B: Booting the last SD Card on the Pi400 before putting it in the Pi 4b was the trick - Ok, tell me why that would be the fix because it booted fine on the Pi 4b - It was the WIFI that kept dropping constantly no matter what the F I did 2 SD Cards in a row! 32 & 64 Bit, Newest Raspberry Pi OS!

- This is exactly why AI needs a kill switch. The impossible is possible. When talking about code at least.

lurch, 2022-04-06:
> It was the freaking Raspberry Pi Imager 1.7.1 upgraded to 1.7.2 without uninstalling the old version first that was the problem!

Nope, I'm fairly sure that wasn't the problem. RPi Imager either works and flashes an OS that boots (because it always verifies the data that it writes to the SD card), or it doesn't work and results in an SD card that doesn't boot. It's not possible for RPi Imager to "bug out the wifi".

> Booting the last SD Card on the Pi400 before putting it in the Pi 4b was the trick - Ok, tell me why that would be the fix because it booted fine on the Pi 4b

There was a problem with wifi connections on the RPi 4B recently, so maybe that's what you were seeing https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/217
IIRC the Pi400 uses a slightly different wireless chip, so perhaps the Pi400 was successfully able to update the wifi driver to a newer working version. :man_shrugging: 

Nothing to do with "AI" :robot:

seaniedan, 2022-04-26:
In case it's relevant: I burned Raspberry Pi OS (64-bit) desktop image with the rpi imager, and specified a name and **weak** password. When I tried to SSH into the machine, I was unable. I got 'Permission denied, please try again'. On the pi itself, I tried to change the password with 'passwd' and it asked for my existing password, however it wouldn't accept it. I got:

`passwd: Authentication token manipulation error
`
..basically there was no password set, or the password was mangled.

maxnet, 2022-04-26:
>In case it's relevant: I burned Raspberry Pi OS (64-bit) desktop image with the rpi imager, and specified a name and weak 
>password. When I tried to SSH into the machine

Weak password does not matter.
We do not enforce password complexity.

How reproducible is the issue?
As in: are you 200% sure you didn't make a typo in the password when setting it...

tiagomaricate, 2022-06-05:
Same here. Also for some reason, 50% of the time I burn an image to an SD card, even the admin pass is saved in the RPI imager, when I try to login I get the message my pass is wrong. So I have to burn it all again.

Looks like the pass we set in the RPI Imager only works for the first burn/mount. If you need to burn/mount another image, you have to set the pass again, even if in the text field you already have the pass used before..

zzaappedd, 2022-12-08:
has happened twice. I suspect the first burn is correct then the next one uses the hash or something that is not the password. I tried the sha256'd pwd but no go. Is there salt? or a different hashing algo?

maxnet, 2022-12-09:
>Is there salt? or a different hashing algo?

Hashing algorithm is the same.
There do is salt.
So if you save the password twice, you get two different sha256crypt values. But the same password lets you in with both.

RaistlinD2x, 2023-03-08:
I'm having this issue. The network password changes into what I'm guessing is a hash. The Pi doesn't show up on my network and when I attempt to log in locally to the device using the username and password from advanced settings it fails. I've tried 3 times so I'm guessing it's unlikely typos.
[image]

Abm116, 2023-11-28:
I'm having the same issue

maxnet, 2023-12-07:
>Same issue with 1.8.1 here too. 

Newest is 1.8.3, available on the github download page.

Also note that in the 2023-12-05 RPI OS Lite image the wifi customization is broken.
Full RPI OS and Ubuntu Server do should work.

lurch, 2023-12-23:
Possibly another duplicate of #531 ?

MechanicalNecro, 2023-12-25:
> Possibly another duplicate of #531 ?

Just got a pi4 today, fresh install of 1.8.4 and seem to be getting the same issue. Going to try again tomorrow after I've had some sleep.

joeyp2011, 2023-12-28:
same issue, annoying as fuck.

also having problems with properly setting up raspi os on my pi 4b and this issue isnt helping

tdewey-rpi, 2024-02-06:
Closing this one as stale - no new reports since December. #531 was the likely cause of the original report, and I'd encourage people to confirm they still see it with 1.8.5 before raising a new issue.

closed 2024-02-06 by tdewey-rpi

SolberLight, 2024-02-27:
Still having the issue on 1.8.5 when using images for Pi Zero v1

lurch, 2024-02-27:
@SolberLight Could you please create a new issue, giving as many details as possible, what steps you performed, and how it's "going wrong"? Thanks.

tdewey-rpi, 2024-02-27:
Locking this issue. As @lurch points out - resurrecting a stale issue is not going to get me to look at a problem - but an issue report where you include the details required by the issue form will.