raspberrypi/rpi-imager #407: Writing failed while trying to write openwrt

buejoulmnq489, 2022-03-27:
Hello,
I am trying to write openwrt to a brand new 32gb sandisk micro sd card but i keep getting an error that says, "Write failed. Contents of SD card is different from what was written to it." What do I do?

maxnet, 2022-03-27:
What is the file size of the custom image you are trying to write?
A number that is not dividable by 512?

buejoulmnq489, 2022-03-28:
im using this file
[openwrt-21.02.2-bcm27xx-bcm2711-rpi-4-squashfs-factory.img.gz](https://github.com/raspberrypi/rpi-imager/files/8363234/openwrt-21.02.2-bcm27xx-bcm2711-rpi-4-squashfs-factory.img.gz)

maxnet, 2022-03-28:
As expected, that is a broken image file.
It is 80346522 bytes uncompressed. 
Disks are supposed to be made up out of a number of 512 byte sectors, so 80346522 byte disks do not exist, as it is a number not dividable by 512.

If you still want to try writing it, you can try the build here that tries to fix up (pad with zeroes) broken images: https://github.com/raspberrypi/rpi-imager/issues/364#issuecomment-1041428869

buejoulmnq489, 2022-03-28:
Ok, do you want me to use that imager for openwrt? Or is the actual openwrt file broken?

buejoulmnq489, 2022-03-28:
Also im a bit confused on what exactly im supposed to do?

maxnet, 2022-03-28:
Either:

1) Point out to the openwrt folks that there openwrt image is broken and ask if they fix their image.
2) Try if the image works with the Imager version linked, that can attempt to write broken images.

buejoulmnq489, 2022-03-28:
Ok. Thank you for thie info.

buejoulmnq489, 2022-03-28:
Ok! The imager wrote openwrt to the sd card. Thank you for your help!

closed 2022-03-28 by buejoulmnq489