raspberrypi/rpi-imager #409: Add openSUSE Images

coogor, 2022-03-29:
The openSUSE Project (https://www.opensuse.org) provides ready-to-run images for the ARM platform, e.g. 
Raspberry3: https://en.opensuse.org/HCL:Raspberry_Pi3
Raspberry4: https://en.opensuse.org/HCL:Raspberry_Pi4

Images are available preconfigured for Leap (classic release model, based on SUSE SLE binaries) and Tumbleweed (Rolling Release) for various desktops: 
JeOS - minimalistic image (Just enough Operating System)
Enlightenment E20
XFCE
LXQT
KDE
X11

Installation images (for boot from USB) are available from get.opensuse.org

What needs to be done to get the openSUSE images into the rpi-imager OS selection dialog?

ghollingworth, 2022-03-30:
You would need to provide a suitable .json file and a top level entry to match others, for example:

http://downloads.raspberrypi.org/os_list_imagingutility_v3.json

Looking at the InfoBeamer entry would show how you'd also be able to provide a weblink for each entry as well.  You can also develop your own sub-menus where you need to have lots of different types of OS / desktop.  Although I'd note that it may become confusing for a user and might be better to choose a single desktop for rpi-imager.

lurch, 2022-03-30:
And note also that the .json file needs to be hosted and maintained by openSUSE themselves, it can't be provided by a 3rd-party.

hellcp, 2022-03-30:
I don't see why we would ask anybody but ourselves to host something like that ;)

tdewey-rpi, 2023-11-17:
Please fill out the form to request an addition to the OS list at https://forms.gle/cbsU27aLLVaxUXbw8.

closed 2023-11-17 by tdewey-rpi