raspberrypi/rpi-imager #410: Test Bootloader

Erebus999, 2022-04-01:
Error downloading the operating system list from the internet.
Error reproducible on 2 clients with different internet access.

maxnet, 2022-04-01:
Any type of proxy server or security software active?

With your normal web browser, you do can reach: https://downloads.raspberrypi.org/os_list_imagingutility_v3.json ?

Erebus999, 2022-04-01:
Yes works.

Test inside the json file:
http://ftp.fau.de/osmc/osmc/download/installers/rpidiskwriterimages.json  -- ok
https://downloads.raspberrypi.org/net_install/beta/eeprom_recovery_beta.json -- Error 403 You don't have permission to access this resource.Server unable to read htaccess file, denying access to be safe

maxnet, 2022-04-01:
Ah, now I understand.
You are not getting the "Error downloading OS list" message on opening Imager when it downloads the main OS list.

But you are getting the error after going to "Misc utility images" -> "Beta test bootloader" because the permissions of https://downloads.raspberrypi.org/net_install/beta/eeprom_recovery_beta.json are wrong.
Yes, I can reproduce that.

@XECDesign can you look at permissions of that file (or its parent directories) on webserver?

mahtDFR, 2022-04-01:
Bump. Also reproduced this on Windows 10

Erebus999, 2022-04-01:
Its works. Thx.

maxnet, 2022-04-02:
Closing as works now

closed 2022-04-02 by maxnet