raspberrypi/rpi-imager #411: wifi check at wtite time

cybercop23, 2022-04-04:
Having the ability to setup the wifi before buring the image is a great help. It will be awesome if the SSID and password could be verified before burning the image to make sure it is valid.
Just tried a valid SSID with the wrong password (v1.7.1) and it didn't complain at all before writting the image.

lurch, 2022-04-04:
I guess that would mean backing up your PC's existing wifi configuration, disconnecting the wifi connection, setting up a new wifi configuration with the details provided to RPi Imager, trying to connect to the wifi to see if it works, disconnecting the temporary wifi connection, removing the temporary wifi configuration, restoring the original wifi configuration, and finally connecting to the original wifi connection.
All sounds quite complicated and potentially error-prone, and I'm sure people would get **very** upset if something went wrong and RPi Imager accidentally "broke" their PC's wifi setup!

maxnet, 2022-04-04:
>I guess that would mean backing up your PC's existing wifi configuration

And that's a problem.
On the Linux platform the normal user that is running Imager generally does not have the right permissions to READ the existing wifi key.
Although -depending on the Linux distribution- a regular user may or may not have the right privileges to SET new wifi access details through DBus.

I do am assuming the TS is using Linux. As on Windows and Mac OS X Imager do SHOULD be able to pre-fill the wifi information from the system, and therefore typing mistakes should normally not occur, unless you want the Pi to use a different network than the computer running Imager.

tdewey-rpi, 2024-02-06:
Rejecting this feature request.

Short of simulating a connection to the network, we have no way to confirm that password is _actually correct_.

closed 2024-02-06 by tdewey-rpi