raspberrypi/rpi-imager #413: Security & Privacy:  - Telemetry: phone back home.

qkum, 2022-04-05:
**Privacy & Informative focused post:**

Most people don't know that "locale" means the name of your PC/Device used to flash the SD card.

- This can pretty easily be tracked back to you, if you never change your device name. 2022.

Quote from the 1.7.0 update:
```
Telemetry: phone back home when image from repository is written:
name and URL of image written, parent category.
about the computer running Imager: OS, version, architecture,
LOCALE, Imager version, Pi revision
```

qkum, 2022-04-05:
What Raspberry Pi needs our Device/PC name for, I would like to know btw. 

It would be much more intelligent to give the App a unique identifying ID that is generated at installation.

- Instead of the Name of our devices.
-  The Device name can be changed Daily if you want to yes. But nobody does that, And even if you did. Apple & Google would still be able to link your device name back to you because every iPhone & Android device scans everything around it 24/7 and reports back to HQ and updates their global WIFI & Bluetooth "triangulation" database.

The Essence:  An unique App ID created at installation would not be in any of their databases (could be hashed) and you would still be able to know if it is the same Device/PC that installed Raspberry Pi OS 10 times in 2 hours, or 10 different people/devices, etc. Detect DDOS attacks etc. (the only legit reason you could, need/ask for,  such private data)

It could change the App ID when the App is re-installed, Updated, or once a month?

Would be a 1000 times better way of doing it anyway.

I hope my rambling can be used for something constructive. 
**Have a nice day**

qkum, 2022-04-05:
I know we don't have `"anything to hide"`

But that does not mean that we should let Big Tech know & track Everything he knows year out and year in so they in the end not just know us better than ourselves but also can predict everything we do,- before we know it ourselves.

_And that day is closer than you think._

cp2004, 2022-04-05:
(disclaimer: not a RPi person, don't know what actually goes on in the RPi imager)

Isn't locale the language/location...? Nothing to do with the device name...

Eg. my locale would be `en_UK`.

qkum, 2022-04-05:
> (disclaimer: not a RPi person, don't know what actually goes on in the RPi imager)
> 
> Isn't locale the language/location...? Nothing to do with the device name...
> 
> Eg. my locale would be `en_GB`.

I came back to correct myself.

I Just fell over the `"Raspberry Pi Imager Statistics"` page just now and saw what "locale" means in this context. 

Here is where I got my assumption of it in this context meaning the Device Name. 
[image]

- 2022 the big guy trends to disguise their dirty tricks behind innocent-looking words - so my lie detector falsely fell over this example.

* Thanks for correcting me. 
* Now people can find the explanation through Google & this thread and save some time and headache.

closed 2022-04-06 by ghollingworth