raspberrypi/rpi-imager #414: Verify passes on sdcards that have failed

guysoft, 2022-04-06:
I tried to flash as a test SD cards that died. When an SD card (aka flash memory) reaches the maximum writes it can do it becomes read-only. This is the kind I am talking about.

I tried to flash such an SD with imager 1.7.1 and it run Verifying and passes. Notably the verify takes much longer.

Steps to reproduce:

1. take an SD card that has died and became read-only
2. Flash any distribution to it from the image
3. Check and see the SD card still has the old content

Expected:
Verify should fail

Actual:
Verify takes much much longer, but passes

maxnet, 2022-04-06:
Which operating system, and if Linux kernel version?

On Linux we do are expecting posix_fadvise() to work as advertised, and discard the block cache:
https://github.com/raspberrypi/rpi-imager/blob/qml/src/downloadthread.cpp#L760
If that for some reason was not working on your Linux kernel version, it may be verifying from memory cache instead of card.

On other operating systems there shouldn't be much of a disk cache.

guysoft, 2022-04-06:
Ubuntu 21.10 kernel:
```
Linux hostname 5.13.5-051305-generic #202107251335 SMP Sun Jul 25 13:38:50 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

lurch, 2022-04-06:
If you try flashing an image that's bigger than your computer's physical RAM, that's one way of doing a test whilst ensuring that the entire image-data isn't being cached :slightly_smiling_face:

guysoft, 2022-04-06:
I have 32GB ram :(

lurch, 2022-04-06:
That must be the first time you've ever been upset about having "too much" RAM :laughing:

maxnet, 2022-04-07:
>If you want me to try some kind of cpp sniplet with posix_fadvise I can try that.

You can typically see it working by running tools like "vmstat 1" which report memory stats in kilobytes to screen.

For me, it does seem to work properly and loses about 4 GB worth of buffers the moment it starts verifying.

[image]

(On KUbuntu 20.4, Linux 5.4.0-104-generic, also with 32 GB memory).

guysoft, 2022-04-07:
I am not sure why we slided from detecting if an sdcard has failed to memory buffers.

Perhaps I can give more info on how an sd card that dies behaves.

It reports successful writes, same for instance if you open it in gparted, it will not complain that it can't write when you, for example, delete all partitions.
However when reading the data you get the old data again. 

This might require a different test.

guysoft, 2022-04-07:
For example here is a solution someone implemented for this:
https://raspberrypi.stackexchange.com/questions/88301/detect-an-sd-card-that-became-read-only

maxnet, 2022-04-07:
>I am not sure why we slided from detecting if an sdcard has failed to memory buffers.

Because when write()'ing to SD card, it not only writes to the SD card, but -if you have plenty of memory- also keeps a copy of everything you write in Linux's block cache.

When read()'ing to verify, it may use the copy from Linux's block cache instead of actually reading from SD card if it is not dropped properly....

>However when reading the data you get the old data again.

The question is if it is actually reading from SD card...

maxnet, 2022-04-07:
>For example here is a solution someone implemented for this:
>https://raspberrypi.stackexchange.com/questions/88301/detect-an-sd-card-that-became-read-only

I do not see any solution there.
Cannot use /proc/sys/vm/drop_caches like suggested there...

```
$ ls -l /proc/sys/vm/drop_caches
--w------- 1 root root 0 Apr  7 13:02 /proc/sys/vm/drop_caches
```

How is Imager being run as a normal user -not as root- going to write to that?

Have to use syscalls that normal users can use. Like posix_fadvise().

guysoft, 2022-04-07:
Thr verify process takes considerably more time when this happens. If the card was in cache, I doubt it would have been slower. 

Ill try find another machine and test. 

Also, I can try dual-boot to windows and test there.

tdewey-rpi, 2024-02-06:
@guysoft Are you still seeing this?

guysoft, 2024-02-06:
The corrupted sd card, which I kept here to test this issue, does not seem to get detected anymore by my linux at all. Even before rpi-imager. So I am not sure I have a way to test anymore. I'll see if I can get a problematic SD card around here.

tdewey-rpi, 2024-08-28:
Closing as 'Stale'.

closed 2024-08-28 by tdewey-rpi