raspberrypi/rpi-imager #416: An advanced option to create the appropriate locale

RanchoHam, 2022-04-09:
While flashing a CM3+ with Bullseye lite, I used the advanced option to set the Time zone and the keyboard layout.

But upon first boot login, I was presented with "**-bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)**".  

To fix this, I had to manually start raspi-config and generate the US.UTF-8 locale.  When I re-logged in everything worked correctly.

I can see that no one wants ALL of the locales to be included in the distribution image, so is there anyway to cause this locale compilation to automatically start on first login?  Some kind of run-once script?

maxnet, 2022-04-09:
Imager only sets keyboard layout and timezone, but not locale.
So could be a RPI OS bug.

lurch, 2022-04-11:
ping @XECDesign

RanchoHam, 2022-04-11:
> Did you log in over SSH? If not, how did the OS know you wanted to use en_US?

It shows up both on the console (I'm flashing Raspberry Pi OS 32-bit) upon auto-login and when logging in over SSH.

XECDesign, 2022-04-11:
Right, but the question is what happened between the time the image was flashed and you saw that message that changed your locale to en_US? The images we ship are pre-configured as en_GB, so if rpi-imager isn't changing the locale, something else is. Whatever that is, it's doing it incorrectly.

RanchoHam, 2022-04-11:
> Imager only sets keyboard layout and timezone, but not locale. So could be a RPI OS bug.

No, I don't think it is an OS bug; because the locale compiler needs to be run to generate the locale files.  Since Imager is only setting the layout and timezone, a way is needed to force the compiler to run upon first boot.  There are some run once schemes floating around the 'net, but I haven't seen one that is particularly elegant.  Perhaps one of the developers knows of one.

RanchoHam, 2022-04-11:
> Right, but the question is what happened between the time the image was flashed and you saw that message that changed your locale to en_US? The images we ship are pre-configured as en_GB, so if rpi-imager isn't changing the locale, something else is. Whatever that is, it's doing it incorrectly.

It appears that that Imager changed the locale environment values, but did not compile the locales to support those variables.

lurch, 2022-04-11:
> It appears that that Imager changed the _locale environment values_, ...

Are you sure? Looking at https://github.com/raspberrypi/rpi-imager/blob/qml/src/OptionsPopup.qml seems to confirm that it's only the keyboard-layout and timezone that are being set by RPi Imager.

maxnet, 2022-04-11:
>No, I don't think it is an OS bug; because the locale compiler needs to be run to generate the locale files.
>Since Imager is only setting the layout and timezone, a way is needed to force the compiler to run upon first boot.

Imager only changes /etc/default/keyboard and /etc/timezone, and should not change the default locale that is en_GB.
And RPI OS does have locale files for en_GB....

E.g. if I chose RPI OS Lite, and set through Imager: custom username/password, "enable SSH", timezone "Europe/Amsterdam", keyboard "us", logging in works fine, and locale shows as en_GB:

```
$ ssh raspberrypi.local
The authenticity of host 'raspberrypi.local (192.168.178.77)' can't be established.
ECDSA key fingerprint is SHA256:lasvkWvecj11AL+JF0/nQrpVQdjSQqMfzNndMuk44N8.
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added 'raspberrypi.local,192.168.178.77' (ECDSA) to the list of known hosts.
Linux raspberrypi 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Apr  4 14:06:36 2022

Wi-Fi is currently blocked by rfkill.
Use raspi-config to set the country before use.

max@raspberrypi:~ $ locale
locale: Cannot set LC_ALL to default locale: No such file or directory
LANG=en_GB.UTF-8
LANGUAGE=
LC_CTYPE="en_GB.UTF-8"
LC_NUMERIC="en_GB.UTF-8"
LC_TIME=nl_NL.UTF-8
LC_COLLATE="en_GB.UTF-8"
LC_MONETARY="en_GB.UTF-8"
LC_MESSAGES="en_GB.UTF-8"
LC_PAPER="en_GB.UTF-8"
LC_NAME="en_GB.UTF-8"
LC_ADDRESS="en_GB.UTF-8"
LC_TELEPHONE="en_GB.UTF-8"
LC_MEASUREMENT="en_GB.UTF-8"
LC_IDENTIFICATION="en_GB.UTF-8"
LC_ALL=
max@raspberrypi:~ $ cat /etc/timezone
Europe/Amsterdam
max@raspberrypi:~ $ cat /etc/default/keyboard 
XKBMODEL="pc105"
XKBLAYOUT="us"
XKBVARIANT=""
XKBOPTIONS=""

BACKSPACE="guess"
```

The question is, how did locale end up being set en_US in your case?

maxnet, 2022-04-11:
Hmm, thinking about it.
It seems on Linux the ssh client takes over locale settings from the computer running the ssh client.

If I set an LC_ALL environment variable on my normal Linux computer to en_US, and ssh to the raspberry, I do can reproduce the TS's problem.

```
$ LC_ALL=en_US.UTF-8 ssh raspberrypi.local
Linux raspberrypi 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Tue Apr 12 01:26:22 2022 from 192.168.178.146
-bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)

Wi-Fi is currently blocked by rfkill.
Use raspi-config to set the country before use.
```

Anyway, not an Imager problem.
But perhaps RPI OS do should generate all common locales to combat this?

RanchoHam, 2022-04-11:
> Hmm, thinking about it. It seems on Linux the ssh client passes through locale settings from the computer running the ssh client.
> 
> If I set an LC_ALL environment variable on my normal Linux computer to en_US, and ssh to the raspberry, I do can reproduce the TS's problem.
> 
> ```
> $ LC_ALL=en_US.UTF-8 ssh raspberrypi.local
> Linux raspberrypi 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l
> 
> The programs included with the Debian GNU/Linux system are free software;
> the exact distribution terms for each program are described in the
> individual files in /usr/share/doc/*/copyright.
> 
> Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
> permitted by applicable law.
> Last login: Tue Apr 12 01:26:22 2022 from 192.168.178.146
> -bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
> 
> Wi-Fi is currently blocked by rfkill.
> Use raspi-config to set the country before use.
> ```
> 
> Anyway, not an Imager problem. But perhaps RPI OS do should generate all common locales to combat this?

Great Guhu, **NO**!  Generating all the locales takes up a bunch of space on an already crowded basic installation.  I'd rather put up with manually running raspi-config once, than add all that dead weight.  I've found a way to script raspi-config, so it becomes a matter of popping in a thumb drive with a shell script or opting for one of the run-once schemes out on the 'net.

maxnet, 2022-04-12:
>Great Guhu, NO! Generating all the locales takes up a bunch of space on an already crowded basic installation.

Yes, adding the top 10 locales (as reported by rpi-imager-stats) do would add 7 MB to the footprint of RPI OS Lite:

```
max@raspberrypi:/usr/lib/locale $ ls -l
total 2976
drwxr-xr-x 3 root root    4096 Apr  4 12:47 C.UTF-8
-rw-r--r-- 1 root root 3041504 Apr  4 12:47 locale-archive
max@raspberrypi:/ $ sudo vi /etc/locale.gen 
max@raspberrypi:/ $ sudo locale-gen
Generating locales (this might take a while)...
  de_DE.UTF-8... done
  en_AU.UTF-8... done
  en_GB.UTF-8... done
  en_IN.UTF-8... done
  en_US.UTF-8... done
  es_ES.UTF-8... done
  fr_FR.UTF-8... done
  ja_JP.UTF-8... done
  ru_RU.UTF-8... done
  zh_CN.UTF-8... done
Generation complete.
max@raspberrypi:/ $ ls -l /usr/lib/locale
total 9944
drwxr-xr-x 3 root root     4096 Apr  4 12:47 C.UTF-8
-rw-r--r-- 1 root root 10223808 Apr 12 01:23 locale-archive
```

Anyway, since such bloat is clearly not wanted, and it is not an Imager specific problem, I am closing this one.

closed 2022-04-12 by maxnet