raspberrypi/rpi-imager #417: Add file open dialog for SSH public key

CharlesGodwin, 2022-04-10:
Issue #326 talks about a request existing somewhere to allow file open to select public key. I searched and couldn't find it. I'd like this too.
I suggest a file open dialog box and you can point on the .pub file, or the private key file, and the imager would copy and paste the public key.

This is a lot simpler than copy and past of the printed key.

maxnet, 2022-04-10:
>This is a lot simpler than copy and past of the printed key.

Note that doing so is only needed if you are using a non-standard SSH client.

Imager puts the contents of ~/.ssh/id_rsa.pub in the field automatically, which is the default location where ssh-keygen stores the public key.

CharlesGodwin, 2022-04-10:
I know that but I, and I'm sure there are others, use more than one key, depending on the device, that's when point and click would help.

This is an ask, not a must do.

tdewey-rpi, 2024-02-06:
Tagging for 2.0.

Identity selection is a complicated topic - it's not even clear we should always use a public key as stored on disk, over for example one kept in Secretive on macOS.

I can't say we'll look at it imminently, but I'm going to keep it open for consideration in the 2.x series of releases.

closed 2025-12-12 by tdewey-rpi