raspberrypi/rpi-imager #418: Add a comment (README) to boot directory

CharlesGodwin, 2022-04-10:
I create a lot of images and if they aren't installed in a working Pi, I have no idea what they are. 

Could you add another option to the imager that would be a free form text area style. Then, if valued, the contents could be put in a file on the boot directory (perhaps README), or added as a comment block to /boot/config.txt

The boot directory is easily read by Windows and Mac OS machines.

I thought of suggesting key fields from imager options to add automatically but that could be fraught with debate, so just make it free form. User's choice.

CharlesGodwin, 2022-04-11:
Please don't get me wrong. I am very grateful for the Pi Imager as it has significantly simplified setting up an image. I have memories, not fond memories, of having to add a ssh file, setup wireless, then use raspi-config to assign host and passwords, then copy public keys, etc. before I even started to setup up the image to do anything.

But, I prefer "one stop shopping"  ;) 

How about this?: Only a thought. <boot> is the boot image name for the destitution. I know not all are /boot
Create a folder` /<boot>/pi-imager/`, then dump this README in there, out of harms way.  And/or combine this with #419 and just dump a copy of the options into this folder. That may be enough.

"No" is an acceptable answer. ;)

DAveShillito, 2022-05-01:
I have been thinking of making the same request for months and since I am in the process of writing a number of SD cards finally decided to come here to make the request, only to find @CharlesGodwin had beaten me to it.

I too have so many SD cards sitting around which are no longer in Pi's and I frequently wonder why I created them. Being able to pop them in a PC and read a comment I added when I initially wrote the SD would be so handy.

I'd also like to suggest the creation date, Linux version and which image file was used as the initial image be added to the text file by default.

CharlesGodwin, 2022-05-01:
My short term solution is to add it myself. I always have a script I run to add required modules etc so I've added a step to stick a file in /boot/notes directory.

tdewey-rpi, 2024-08-28:
Feature request rejected.

I sympathise with the use case - however I can't see this feature being particularly widely used, and the workaround is very straightforward - just insert your notes manually, and ensure you follow a convention to easily find them in every card you write.

closed 2024-08-28 by tdewey-rpi

lurch, 2024-09-03:
I can also sympathise (I have lots of different SD cards that I use for testing purposes), but I ended up solving this in a slightly different way. I have a unique-numbered-sticker on each SD card, and then a spreadsheet on my laptop where I record which image was written to which card, along with a brief description. I also use the same spreadsheet to record the username and password I setup on each card (which obviously isn't the best security practice, but these are effectively "throwaway images" which usually only last for a couple of weeks before getting overwritten with something else).

CharlesGodwin, 2024-09-03:
I tried your idea but lacked the discipline to do it right. Now I just add this script, suitably revised, to every pi setup. It creates a snapshot at every boot and if anything changed, updates a text file on the /boot folder and keeps a numbered backup as well.
[pi-setup.txt](https://github.com/user-attachments/files/16849982/pi-setup.txt)

DAveShillito, 2024-09-04:
Likewise, in the 2 years since this was suggested I have never remembered to add a note before I eject an imaged disk.
That's why I like these things to be automated.

I understand this issue has been closed, but an alternative suggestion would be to allow a command to be specified which would be run on completion of the write.  This could be used to run a script such as the one @CharlesGodwin suggested, or to fire a webhook to alert the user that the process is complete by flashing house lights etc. You could even build a robot arm to remove the SD and replace it to make an automated SD imaging factory :) 
I can see that this would have a much wider appeal above that initially suggested.

If variables such as the date, sd drive path and img filename could be passed as command parameters I can see many uses of such a feature.

lurch, 2024-09-04:
If you want to add functionality to RPi Imager to flash your house lights on and off, you're free to do so; it's all open-source software. :bulb: :flashlight: :house: 

But as @tdewey-rpi has mentioned in other issues, the primary target for Raspberry Pi Imager is beginner-users, and so I doubt we'll be including features to allow users to run arbitrary scripts. See https://github.com/raspberrypi/rpi-imager/issues/837#issuecomment-2006625835 for more details on rationale.

CharlesGodwin, 2024-09-04:
I considered a timestamp in the contents but then every boot would be a different file. i rely on the file date for the date info.

DAveShillito, 2024-09-04:
TBH, I never really considered that RPi Imager was a tool for just beginners since it is the best tool for everyone to use IMHO.

Adding this feature to the software myself was added to my todo list a long time ago... but it is itself a very long list, and I didn't want to add learning QT to it as well. ;)

However since this subject from the past has been re-animated (ironically by closing it) I was wondering if I could use the CLI to wrap the entire process in my own interface. However it appears the CLI does not expose all the RPi Imager secret sauce such as setting the username/wifi/password etc. :(

Ah well, it was only a suggestion, I shall move on.
Thanks for the feedback :)

lurch, 2024-09-04:
> since it is the best tool for everyone to use IMHO.

I of course agree :wink: but we have to be careful not to let feature-creep get out of hand, and let RPi Imager become so complicated that it's no longer suitable for beginners because it's too overwhelming.

CharlesGodwin, 2024-09-04:
I'm OK with it as it is. Feature creep is always a problem

CharlesGodwin, 2024-09-06:
I had a thought. I know it's feature creep. Instead of this idea of a special comments file etc., why not provide an ability to run a user supplied script as part of the initial boot process. This script could do anything asked of it and is user defined.

My first thoughts are:

- Add prompt for script file name to imager warning user that the quality of the script is a user problem
- copy the file to the boot folder in the image
- after all else has run during initial boot(s) run the script in the defined use space.

In my case of wanting a comments file, I'd provide a script that would build that at run time. (I can provide a sample of the script script I would use, if you want to see it).

What do you think? No is an acceptable answer. ;)

lurch, 2024-09-06:
> I had a thought. I know it's feature creep. Instead of this idea of a special comments file etc., why not provide an ability to run a user supplied script as part of the initial boot process. This script could do anything asked of it and is user defined.

That's already been requested (and rejected) multiple times :slightly_smiling_face: 
See e.g. https://github.com/raspberrypi/rpi-imager/issues?q=is%3Aissue+firstboot

CharlesGodwin, 2024-09-06:
Pity. Thanks for the follow-up. I'll stop posting.

lurch, 2024-09-06:
> I'll stop posting.

Sorry, I hope I've not accidentally offended you?
We **do** welcome comments and feedback on Raspberry Pi Imager, but we're very careful with which feature-requests we do or don't accept.

Note also that Raspberry Pi Imager tries to be OS-agnostic, and so even if we _did_ offer a feature to copy user-defined scripts to the boot partition, this likely wouldn't work with many of the OSes offered by Imager, which would (inevitably) lead to user-confusion.

CharlesGodwin, 2024-09-06:
No, I m not offended in any way! It's just that this a closed issue and I think I've hit the end of the road for possible enhancement. I'm not mad as I already have an alternative solution that works well enough.

Thank you for your concern.