raspberrypi/rpi-imager #419: Add save/load of imager options

CharlesGodwin, 2022-04-10:
I manage my image setup in a git repository so I can readily check the repository to determine what the setup for a particular device (pi) is. I use scripts to do initial setup such as adding packages etc. This make build a new image reasonably easy and error free.

It would be nice if I could save the options used to create a certain image, archive it with git, and then load it if I need to create a new copy of the image.

I think several items should not be saved, such as the device to write image and any passwords or authentication keys.

maxnet, 2022-04-10:
>I use scripts to do initial setup such as adding packages etc

May be better to clone https://github.com/RPi-Distro/pi-gen for that.

CharlesGodwin, 2022-04-10:
There are two problems with pi-gen suggestion.
1 I prefer to use a solution under the care of the Raspberry pi team
2 I don't use Linux as my workstation so pi-gen won't work for me

maxnet, 2022-04-10:
>1 I prefer to use a solution under the care of the Raspberry pi team

The "raspberrypi", "RPi-Distro" and "raspberrypi-ui" github accounts, all belong to Raspberry Pi Ltd.

CharlesGodwin, 2022-04-10:
Good to know

Charles Godwin

lurch, 2022-04-11:
Indeed, see https://raspberrypi.github.io/

CharlesGodwin, 2022-04-11:
@maxnet I studied this "May be better to clone https://github.com/RPi-Distro/pi-gen for that."

I think the pi-gen tool is overkill for what I do. All I'm doing is using a script to setup the modules and python packages I need for this image. I do this so I can replicate it easily when the image media (AKA SD card etc.) fails or Raspberry Pi upgrades Debian to 'bookworm" and beyond. I can build a new one in a few minutes. I think pi-gen would be great if I was in the image cloning business.

BTW I keep generated data separate and backed up.

CharlesGodwin, 2022-04-11:
yes they do. I'm considering ansible. The thread was about saving, in text, the options used for this image so they can be archived and loaded when needed.

maxnet, 2022-04-12:
>The thread was about saving, in text, the options used for this image so they can be 
>archived and loaded when needed.

But you didn't want credentials in there.
So essentially what official Imager options are left? Hostname, keyboard layout and timezone?

Sounds a bit overkill to me to design Imager functionality to load and save just that...

tdewey-rpi, 2024-02-07:
Feature request rejected.

Per @maxnet's final comment, this would pose an undue burden on rpi-imager to implement, but there's nothing stopping you saving the rpi-imager settings data outside of the program - if you look at where QSettings saves the options on your platform of choice, you should be able to create a system for loading profiles of options.

That said, I must again strongly recommend you use pi-gen if you're generating a common image (even a standard image with common configuration!) for multiple Raspberry Pi devices. I like to think of the process of creating a device as (at least) two stages:

1. Reference Image Creation
2. Device Provisioning

rpi-imager is very firmly aimed at (2), and pi-gen aimed at (1). The OP's description of their use case sounds like they actually want both.

closed 2024-02-07 by tdewey-rpi

lurch, 2024-02-07:
The solution is obvious: "Raspberry PiGenmager" ! :rofl: