raspberrypi/rpi-imager #420: Missing piwiz suppression?

EternityForest, 2022-04-10:
I see a recent commit: "Remove overscan/piwiz supression advanced options", but no explanation for why it was removed.  

This was a pretty useful feature for digital signage, is there a suggested upgrade path for this, and if so, should it be documented?

lurch, 2022-04-11:
That's probably related to https://www.raspberrypi.com/news/raspberry-pi-bullseye-update-april-2022/ ?

EternityForest, 2022-04-11:
I'm not entirely sure I see the connection.   You can set up the
user/password directly in the imaging utility, doesn't that make piwiz
optional, if you preconfigure that way?

lurch, 2022-04-11:
I've not experimented with it myself, but I _think_ if you configure things using RPi Imager, then it automatically skips the piwiz step? ping @spl237 for more details.

XECDesign, 2022-04-11:
Overscan actually hasn't worked for quite a while.

The piwiz option was removed because it didn't work with the new setup. If you use the option with a new desktop image, you'll not end up with a working desktop environment.

piwiz won't run if you set a username and password through rpi-imager or a userconf file.

tdewey-rpi, 2024-02-07:
Explanation provided by @XECDesign in '22. Closing.

closed 2024-02-07 by tdewey-rpi