raspberrypi/rpi-imager #421: Is the telemetry collected compliant with PECR / EU Privacy Directive?

markalanrichards, 2022-04-10:
I noticed https://github.com/raspberrypi/rpi-imager/issues/391

You may want to check whether you are compliant with PECR - the EU Privacy Directive which has some coverage for the protection of not just personal data when collecting data from a users device (terminal). 

I'm no legal expert, but my reading of https://ico.org.uk/for-organisations/guide-to-pecr/cookies-and-similar-technologies/ suggests that this warrants a check to be safe.

PECR / EU Privacy Directive is commonly known as the cookie law, but has a much broader scope regarding the privacy of various types of data on a users device (not just personal data)
> What else is covered, apart from cookies?
> Although this guide focuses on cookies, regulation 6 actually applies to anyone who stores information on a user’s device or gains access to information on a user’s device, in either case by any method.

memjr, 2022-05-05:
The telemetry collected by the pi, if you choose to keep it enabled (you can disabled it in the imager before you burn an image), does not collected any personal information. Only things like which OS you're running, whether it's 32 or 64 bit and the like.

memjr, 2022-05-05:
> I noticed #391
>
> You may want to check whether you are compliant with PECR - the EU Privacy Directive which has some coverage for the protection of not just personal data when collecting data from a [...]

The telemetry collected by the pi, if you choose to keep it enabled (you can disabled it in the imager before you burn an image), does not collected any personal information. Only things like which OS you're running, whether it's 32 or 64 bit and the like.

markalanrichards, 2022-05-15:
Thank you replying @memjr 

However, where you state _does not collected any personal information_
The ICO guidance also states

_Although cookies that process personal data give rise to greater privacy and security risks than those that process anonymous data, PECR apply to all cookies._

I am not a legal expert, but my hope is that the intent of this legislation goes beyond personal data.

**TL;DR - if you are curious why I hope the legislation does this then read on, but essentially the electronic data on our devices should be protected regardless of whether it is associated to personal information or not.**

Taken to the extreme, were someone in the military and had missile codes on their device, then GDPR would not protect that data from being captured by installed software, but a general law on electronic privacy would.

Far more likely, some of us have shared api tokens to web servers, licence keys to software or even commercial intellectual property on our devices that we must keep confidential and I would hope PECR protects us from companies that develop software we install on our devices, from slurping up any of that data for their own purposes.

Capturing data about what kind of operating system I run isn't the worst thing in the world, but were I running Red Star OS or using an out of support Linux distro (my device is vulnerable to attacks) then I probably would not want any software to needlessly capture that.

aallan, 2022-05-16:
> Is the telemetry collected compliant with PECR / EU Privacy Directive?

Yes.

markalanrichards, 2022-05-17:
@aallan 
Thanks for your reply, however when I read the following I'm not sure why it is a yes. 
Would you mind elaborating on why this telemetry is exempt?

 https://ico.org.uk/for-organisations/guide-to-pecr/guidance-on-the-use-of-cookies-and-similar-technologies/how-do-we-comply-with-the-cookie-rules/#comply15 

_Are analytics cookies exempt?
No. It is important to note that PECR does not distinguish between cookies used for analytics activities and those used for other purposes. Analytics cookies do not fall within the ‘strictly necessary’ exemption. This means you need to tell people about analytics cookies and gain consent for their use.

..._

tdewey-rpi, 2024-08-28:
Closing and locking.

Closing as this is not an issue concerning the functionality or use of `rpi-imager`, and should really have been directed at the Data Protection contact for Raspberry Pi, which is listed on the Raspberry Pi Privacy Policy - along with escalation steps if you are unhappy with the response.

Locking, because I do not see that further discussion on this issue tracker will have any value.

closed 2024-08-28 by tdewey-rpi