raspberrypi/rpi-imager #422: Imager 1

Martin-HiPi, 2022-04-11:


maxnet, 2022-04-11:
>I'm creating a Bootloader Recovery card

Is that an entry from the menu, or with "use custom"?

Martin-HiPi, 2022-04-11:
Sorry, I should have mentioned using Misc utility images -> Bootloader -> (SD Card Boot is the one I used)

When using "Custom", I usually uncheck the Advanced options for incompatible images, but using the above option, the Advanced options don't show (no gear icon).

MLambi-Dover, 2022-04-12:
I had the same error, but my SD card came out without even a /boot on it.
I used the Raspberry Pi Imager v1.7.2, choose the default option of Desktop, and added customer settings (hostname, WiFi, ssh, password).

I tried a second time with an SD card that already had the /boot partition, and it worked.  I am thinking that the first time around the partition was not created.

lurch, 2022-04-12:
Related issues: https://github.com/raspberrypi/rpi-imager/search?q=Unable+to+customize&type=issues
(in case anything there helps you track down / eliminate the cause of the issue)

maxnet, 2022-04-12:
>Related issues: https://github.com/raspberrypi/rpi-imager/search?q=Unable+to+customize&type=issues

Are not related issues.
With the other issues the users do want customization to happen, but there is something wrong with writing or mounting the FAT partition.

While with the EEPROM recovery case no customization should be happening. But somehow the code does think it should, and thus expects to find config.txt/cmdline.txt on the boot partition, while those are not present with the EEPROM recovery.bin stuff.

lurch, 2022-04-12:
Sorry @maxnet , I guess I should have clarified that my comment was directed at @MLambi-Dover (who said "and added customer settings").

thedevopscoder, 2022-05-14:
How to resolve the issue.
If failed open disk management and assign the system-boot portion a drive letter. Re-run wizard and this should write with no issues.

The next time Pi Imager runs it will assign a drive.

tdewey-rpi, 2024-02-07:
Excellent sleuthing, @thedevopscoder.

I'm closing this issue, as it's not directly related to the rpi-imager codebase.

closed 2024-02-07 by tdewey-rpi