raspberrypi/rpi-imager #425: Since 1.7.2 - <Drivce-Letter>:/config.txt does not exist

Marsupilani, 2022-04-15:
Since update to 1.7.2 I am unable to customize the image. Auto eject is disabled. I got the message, that e:/config.txt does not exist at the end of imaging. Windows 10 pops up to want to format the SD-Card. The image on the sd-card is runnable, but requested for new username and Wi-Fi settings are not set.
I set username(pi)/password;Wifi-Country/keyboard layout/Wi-Fi-Setting before under the advanced settings. Image raspberry os lite. until 1.7.1 this settings worked (with older raspberry os lite (bullseye) image). maybe a problem with new forced username change?
I successed to implement offering a working headless installation mode for the lite image (maybe with throwing up a temporary AP like tasmota)...

lurch, 2022-04-15:
> I successed to implement offering a working headless installation mode for the lite image (maybe with throwing up a temporary AP like tasmota)...

I afraid I can't tell what you're trying to say, but https://www.raspberrypi.com/documentation/computers/configuration.html#setting-up-a-headless-raspberry-pi explains how to configure the user in a headless setup.

Marsupilani, 2022-04-16:
The problem is, that after flashing the image, the boot partition of the SD card no longer available for windows (it seems to be still a fat32 partition) but windows does not find the partition, so I am not able to make the changes you explain in your link from a windows 10 client anymore.

lurch, 2022-04-16:
With some SD-readers, you sometimes need to physically remove and re-insert the SD-card for Windows to re-read the new partition table? (If Windows is still "seeing" the old partition table, then it won't see the new boot partition)

closed 2024-08-28 by tdewey-rpi