raspberrypi/rpi-imager #426: Repeated "Verifying write failed" Version 1.7.2

M1DLG, 2022-04-19:
I am getting a verify failure on all of my (branded) cards (dozens) several different writers. I can't boot them when they fail. I have no permissions issues, AV software blocking card access, of counterfeit cards, all SD cards are genuine from reputable sources.
Edition	Windows 10 Pro
Version	21H2
Installed on	‎18/‎02/‎2021
OS build	19044.1645
Experience	Windows Feature Experience Pack 120.2212.4170.0
Imager Version 1.7.2

M1DLG, 2022-04-20:
A follow-up to this issue. Clearing the cache that contained the files downloaded prior to upgrading to 1.7.2 and switching to a USB 2.0 Card reader worked. Not sure what the reason for the repeated verify fail (and unbootable SDcard) was.

lurch, 2022-04-20:
It may or my not be related, but there's also a report [here](https://github.com/balena-io/etcher/issues/3399#issuecomment-906545519) of Etcher failing with a USB3.1 card-reader but working with a USB2.0 card-reader.

@M1DLG If you have a USB2 hub that you're able to plug your (previously-failing) USB3 card-readers into (which will "downgrade" the card-readers to USB2), that might be an interesting experiment?

tdewey-rpi, 2024-08-28:
Closing as 'Stale'

No activity in over 2 years.

closed 2024-08-28 by tdewey-rpi