raspberrypi/rpi-imager #427: Include Debian as installable OS

Fraggle2, 2022-04-24:
I would like too see Debian included in the list of supported OSes. While the Debian project provides no official installation media yet, the OS supports the various Raspberry PI models well (both 32 and 64 bit) and a Debian developer offers semi-official images at https://raspi.debian.net/.
As you can see at https://salsa.debian.org/raspi-team/image-specs/-/issues/60, the developer is open-minded to this idea but lacks time to push this forward himself. If you give me some starting hints, I would like to offer my help to get Debian included into the rpi-imager.

ghollingworth, 2022-04-25:
To add Debian, you just need to provide a .JSON file and top level section to go into the top level JSON file.

See the top level JSON site here:

http://downloads.raspberrypi.org/os_list_imagingutility_v3.json

You should be able to use the Ubuntu one for reference.  Also note you can add a website link to rpi-imager so users can click to find out more information about the build.

Please make sure the Debian .JSON file is then hosted from debian.net somewhere (so we can use https to avoid MITM attacks)

Thanks

Gordon

ghollingworth, 2022-04-25:
To test it, you can run rpi-imager from the command line with the --repo option to use a local file

rpi-imager --repo my_test.json

tdewey-rpi, 2023-11-17:
Please fill out the form to request an addition to the OS list at https://forms.gle/cbsU27aLLVaxUXbw8.

closed 2023-11-17 by tdewey-rpi