raspberrypi/rpi-imager #428: 1.7.2: Pi OS Lite requires "at least 2.0GB" media, but won't allow write to 2GB card

srcshelton, 2022-04-26:
I suspect there's a rounding issue here, but the 297MB Pi OS Lite (32bit) image is reported as:

```
Needs to be at least 2.0 GB
```

… but attempting to write to a 2GB/1.8GiB/3862528 sector SD card results in the error:

```
Storage capacity is not large enough
```

srcshelton, 2022-04-26:
(… although that 297MB is compressed, of course!)

Yeah, so it looks as if 2GB/1.8GiB media is 69632 sectors (~34MB) too small for the minimal image :(

So it'd be nice if the image could be trimmed to make it fit… but I still think there's a bug here, in that the Imager should either say that 2_GiB_ media is required (and use GiB elsewhere for consistency!) or the error message should report that 2.15GB is required, so save exactly this confusion!

lurch, 2022-04-26:
Duplicate of #354 ?

srcshelton, 2022-04-27:
> Duplicate of #354 ?

Yeah, apologies - I did search for a pre-existing issue, but didn't find anything.

(Of note, there's only actually ~1.2G of data on the Pi OS Lite image, it's just the image file-system's free space which prevents it from fitting onto 2GB media!)

lurch, 2022-04-27:
ping @XECDesign in case he wants to offer any further clarifications.

tdewey-rpi, 2025-12-12:
This has been reworked in the 2.0-series.