raspberrypi/rpi-imager #429: Deprecated libssl1.1  in deb

LyzardKing, 2022-04-27:
The .deb file downloaded from the repo has as a dependency the libssl1.1  package.
Since it is not available any more in the ubuntu 22.04 repos it should be replaced with libssl3

maxnet, 2022-06-03:
Problem is libssl3 is in turn not available in earlier Ubuntu versions than 22.04....

LyzardKing, 2022-06-03:
then would it be possible to depend on either one of the two?
If the changes to libssl do not permit this it might be better to build a specific version for ubuntu22.04 and newer

maxnet, 2022-06-03:
We are currently only using libssl for SHA256 computation, as it offers nice hardware accelerated hash routines (e.g. using the special CPU instructions Intel CPUs have for that).

May have a look if another library like libgcrypt has hardware acceleration as well. But will need some investigation.

LyzardKing, 2022-06-03:
I wonder if Qt has a library that can import the proper version of libssl depending on the one available on the system.
I assume that the versions of qt available in the ubuntu repos have been built on the different versions of libssl, and if it could pull the proper version automatically.

maxnet, 2022-06-03:
It certainly possible to load libraries dynamically with dlopen(), dlsym() and friends, as well.
But finding the right library file to load is a pain in the @$#.

Finding a library that offers what we want, and does not change ABI every other week, causing Ubuntu to bump version number, etc., may be less pain.

guysoft, 2022-06-14:
Hey here is the output on ubuntu 22.04:
```
$ sudo dpkg -i /tmp/imager_1.7.2_amd64.deb 
Selecting previously unselected package rpi-imager.
(Reading database ... 268963 files and directories currently installed.)
Preparing to unpack /tmp/imager_1.7.2_amd64.deb ...
Unpacking rpi-imager (1.7.2) ...
dpkg: dependency problems prevent configuration of rpi-imager:
 rpi-imager depends on libssl1.1 (>= 1.1.0); however:
  Package libssl1.1:amd64 is not installed.

dpkg: error processing package rpi-imager (--install):
 dependency problems - leaving unconfigured
Processing triggers for mailcap (3.70+nmu1ubuntu1) ...
Processing triggers for desktop-file-utils (0.26-1ubuntu3) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Errors were encountered while processing:
 rpi-imager
```
Also related it happends on the appimage:
https://github.com/guysoft/pi-imager/issues/10

lurch, 2022-06-14:
@guysoft Does using `apt` instead of `dpkg` help alleviate that problem?

And the AppImage is a 3rd-party package, not directly supported by Raspberry Pi.

maxnet, 2022-06-14:
Can someone that has Ubuntu 22.04 test if the following package does work?
[rpi-imager_1.7.2_amd64.deb](https://www.maxnet.eu/tmp/ubuntu/rpi-imager_1.7.2_amd64.deb)

Outsources SHA256 hash computation to libgnutls instead of libopenssl. Which has a better track record of not changing ABI that often.

LyzardKing, 2022-06-15:
@maxnet It works!

guysoft, 2022-06-15:
Works, confirmed. 

> Can someone that has Ubuntu 22.04 test if the following package does work? [rpi-imager_1.7.2_amd64.deb](https://www.maxnet.eu/tmp/ubuntu/rpi-imager_1.7.2_amd64.deb)
>
> Outsources SHA256 hash [...]

guysoft, 2022-06-15:
Confirming also the deb works both on Ubuntu 21.10 and also 22.04

microhobby, 2022-08-04:
@maxnet from the raspberry pi website `.deb` is still with `libssl1.1`

maxnet, 2022-08-04:
>@maxnet from the raspberry pi website .deb is still with libssl1.1

Correct.
Newer release (with some other minor changes) still needs to pass QA.

maxnet, 2022-09-08:
Fixed in 1.7.3

closed 2022-09-08 by maxnet