raspberrypi/rpi-imager #433: Download images behind a corporate firewall not working

amamabel, 2022-05-05:
On Windows the Pi Imager can't download new images if the PC is behind a corporate firewall. It look like that the Imager does not use the certificate store of Windows.

[image]

maxnet, 2022-05-05:
It do should use the Windows certificate store. The schannel mentioned in the message is the Windows SSL library.

Your message is not about an unknown certificate, but about it unable to check if the certificate used is revoked.
Apparently the certificate used mentions a Certificate Revocation List server that should be consulted to double-check if the certificate has not been revoked, but your firewall has been misconfigured to block access to said server.

lurch, 2022-05-05:
Do you see that error as soon as Imager starts, or only when you try to select an OS?

maxnet, 2022-05-05:
>Do you see that error as soon as Imager starts, or only when you try to select an OS?

Can only occur during Image download.
(Image list download occurs through Qt/qml and that does a bad job checking certificates, but libcurl that is used for image downloads does do it properly)

In the case of downloads.raspberrypi.org the server that normally would be contacted to check if SSL certificate is still valid is: ocsp.sectigo.com
But if behind a corporate firewall that likes to "monitor" employee traffic, the certificate seen by Imager can be provided by the security software instead, and then whatever revocation list server is mentioned in that certificate will be consulted instead.

amamabel, 2022-05-05:
> It do should use the Windows certificate store. The schannel mentioned in the message is the Windows SSL library.
>
> Your message is not about an unknown certificate, but about it unable to check if [...]

If i try to open the mentioned IP in the browser i got a message from our firewall that the IP is blocked. On a second try with the Imager the error message contains another IP. Unfortunately the downloads.raspberrypi.org host is working in my browser, I think our firewall has some problems with the load balancer of downloads.raspberrypi.org. I will redirect the issue to our firewall admin.

Thanks for the quick help :-)

DeadRabbits307, 2022-11-24:
As a workaround for a short term solution: Download manually the image  [here](https://www.raspberrypi.com/software/operating-systems/) and choose as OS an own image located on your local storage

markusschaber, 2023-04-19:
> As a workaround for a short term solution: Download manually the image [here](https://www.raspberrypi.com/software/operating-systems/) and choose as OS an own image located on your local storage

It works, but some extended settings (like preconfiguring host name and ssh certificate) won't apply this way, it seems.

WhiteChamp, 2023-11-23:
> As a workaround for a short term solution: Download manually the image [here](https://www.raspberrypi.com/software/operating-systems/) and choose as OS an own image located on your local storage

how to flash this on sd card?

maxnet, 2023-11-23:
>It works, but some extended settings (like preconfiguring host name and ssh certificate) won't apply this way, it seems.

Using which version of Imager, and which operating system image?
Need at least Imager v1.7.3 to properly customize RPI OS bookworm.

closed 2024-09-18 by tdewey-rpi

markusschaber, 2024-09-19:
@tdewey-rpi The problem is that the affected users (especially in corporate environments) cannot fix it on their side. The correct fix would be for rpi-imager to honor the system certificate store for trusted root certificates.

tdewey-rpi, 2024-09-19:
@markusschaber I got a notification for a comment that appears to have disappeared into the ether, so I'm going to inline it here:

`@tdewey-rpi The problem is that the affected users (especially in corporate environments) cannot fix it on their side. The correct fix would be for rpi-imager to honor the system certificate store for trusted root certificates.`

That is indeed a step `rpi-imager` could take, and one that it has took with 1.9.0.

Keeping this one closed.

amamabel, 2024-09-19:
@tdewey-rpi nothing changed for me with 1.9.0
[image]

tdewey-rpi, 2024-09-19:
> @tdewey-rpi nothing changed for me with 1.9.0 [...]

Please raise a new issue.