raspberrypi/rpi-imager #434: imager not working on OS X

greenonline, 2022-05-06:
When I download the [macOS imager disk image][1] from the [Raspberry Pi OS page][2], and then double click it I get this error:

[![Corrupt image - Finder dialog][3]][3]

If I try to fix it in Disk Utility I get the same error:

[![Corrupt image - Disk Utility dialog][4]][4]

The issue is the same as described here, [imager_1.4 - "corrupt image" on OS X 10.9.5][5], and no solution is provided, other than using `dd`. 

Note that this fails on 10.8.5 and 10.9.5 - I haven't tested on anything newer.

FWIW, I've tried repairing the image in Disk Utility, hence the error shown above, as well as this Q&A on SU, [Unable to open DMG files - "Corrupt Image"][6], which is as equally ineffective.

Does anyone know why the disk image fails to mount? Is there a way of getting it to mount? Or another source of an uncorrupted image?

  [1]: https://downloads.raspberrypi.org/imager/imager_latest.dmg
  [2]: https://www.raspberrypi.com/software/
  [3]: https://i.stack.imgur.com/Bjpb9.png "Corrupt image - Finder dialog"
  [4]: https://i.stack.imgur.com/MIwBA.png "Corrupt image - Disk Utility dialog"
  [5]: https://forums.raspberrypi.com/viewtopic.php?t=283889&sid=18a8b0004e429631ef7a66459befa8e7
  [6]: https://www.youtube.com/watch?v=TggCS8q5yGE

maxnet, 2022-05-06:
>Note that this fails on 10.8.5 and 10.9.5 - I haven't tested on anything newer.

We do not support 9 year old operating systems.
.dmg files have a different format (compression algorithm) nowadays, so that you do not get a more friendly error message that your Mac is too old is also expected.

Absolute minimum is currently Mac OS X 10.13
(And may eventually become 10.14 if we ever switch to universal Intel/M1 builds)

timg236, 2022-05-06:
It works nicely on Mac OS X 10.13, unfortunately there's a lot of macbooks out there that can't be upgraded beyond that because Apple dropped support in newer OS's. Although, there's still the option of building the Imager from source.

maxnet, 2022-05-06:
>Although, there's still the option of building the Imager from source.

Not sure if that would work either.
Even if you compile it against an older version of Qt (that support versions prior to OS X 10.13), chances are you may run into problems with OpenSSL.
Chances are you need an older OpenSSL version, and that may or may not speak a SSL flavour recent enough to be able to communicate with the webservers that host the images...

big-harry, 2022-05-28:
Same issue on 12.3.1

[image]

maxnet, 2022-06-03:
>Same issue on 12.3.1

That does not look like the same error message the TS is having.

Did you tried to download it a second time (may also want to try downloading it from the github releases page here), to make sure nothing went wrong with the download itself?

big-harry, 2022-06-04:
I have, it works when renamed to a .iso

tdewey-rpi, 2024-08-28:
Closing as 'Stale'.

I can confirm v1.8.5 works on current supported macOS versions, and there has been no activity on this issue in over 2 years.

closed 2024-08-28 by tdewey-rpi