raspberrypi/rpi-imager #435: I'd like to use English instead of the language of the OS

Griiid, 2022-05-08:
1. English is friendly for most developers, I hope we can have an setting to choose English or the language of the OS.
2. I'm using zh-TW language, but I'm not sure what file name should I use to add an i18n PR for the language.

maxnet, 2022-05-08:
>English is friendly for most developers, I hope we can have an setting to choose English or the language of the OS.

Try CTRL+SHIFT+L

>I'm using zh-TW language, but I'm not sure what file name should I use to add an i18n PR for the language.

Suspect rpi-imager_zh_TW.ts
But try it first.

Griiid, 2022-05-09:
Great, Ctrl+Shift+L works!
I'll try to translate, thank you.

closed 2022-05-09 by Griiid

JereckNET, 2025-06-30:
Every time I relaunch Imager, the interface is back to the initial language.
Looks like the choice is not persistent.

Is it a bug or a feature ?