raspberrypi/rpi-imager #437: Cannot configure RaspiOS via Imager Tool (v1.7.2) on MacOS 12.3.1

arag0re, 2022-05-10:
I tried to flash raspiOS 64bit lite onto my sd card. Sometimes the flash fails completely sometimes it works but im never able to edit / configure the raspi settings before initial boot via the imager tool.
On Windows everything works as expected but on mac the tools isnt usable rn.
If i can help with something like logs or something else just lemme know.

lurch, 2022-05-10:
Can you provide more details about the SD card-reader you're using? Do you have a different USB card-reader you could test with?

arag0re, 2022-05-10:
I am using the built-in SD Card reader of the [macbook 18,8](https://support.apple.com/en-us/HT204384) from 2021

arag0re, 2022-05-10:
yeah i will try with an external usb sd card reader and let u know how it went

arag0re, 2022-05-12:
So i tried an usb-c microSD Card reader and i was able to flash it. But i got some errors at first.
Also i figured out that, if i check the "Eject media when finished" box, it'll brick the device configuration step of the imager and the tools is not able to change device name or username and password.

maxnet, 2022-06-03:
>So i tried an usb-c microSD Card reader and i was able to flash it. But i got some errors at first.

What were those errors?

arag0re, 2022-06-10:
it says that i cant write to the sd card, after several retries it works
idk its weird behaviour

tdewey-rpi, 2024-08-28:
Closing as 'Stale'. Not activity in >2 years.

closed 2024-08-28 by tdewey-rpi