raspberrypi/rpi-imager #441: Where will the image download to?

sgon00, 2022-05-17:
I am running debian Linux host.
I tried to find where the image will download to by googling. But no luck. 
I also tried to use `find` command to find the `.img.xz` file, but no luck.
Can anyone tell me where will rpi-imager download the image to? What is the default download path?
Thanks a lot.

maxnet, 2022-06-03:
Typically (if not affected by XDG environment variables, etc.) in ~/.cache/Raspberry Pi/Imager/lastdownload.cache

lurch, 2022-06-04:
> Why doesn't the program store all downloaded files instead of only the last one?

Because then people would complain "RPi Imager is eating up all my disk space!".

See also #44, #209, #300 and #322

ffgambler, 2023-10-26:
> in ~/.cache/Raspberry Pi/Imager/lastdownload.cache

So for Raspbian it will be /var/cache, right?
I just have written my first SD card but I can't find the image file in /var/cache/... there is no Imager folder.

maxnet, 2023-10-27:
>So for Raspbian it will be /var/cache, right?

Eh, why would that be?
~/.cache is not the same as /var/cache
~ is a shortcut to your home directory ( /home/something )

ffgambler, 2023-10-27:
Ok, my mistake. Thank You.