raspberrypi/rpi-imager #444: Feature request - passwordless ssh login mode

lattice0, 2022-05-19:
I just setted pubkey login mode and it still allowed password mode. Some people are going to put the password as `raspberry` and still get it insecure. Also I don't want my relatively weak password to be brute forced. Should have an option to disable password SSH login entirely.

Thanks!

lurch, 2022-05-19:
LOL, #342 seems to be asking for the exact opposite? You can't please all the people all the time... :wink:

lattice0, 2022-05-19:
You can, but you have to add the option to make it exclusive one or another

maxnet, 2022-06-03:
>I just setted pubkey login mode and it still allowed password mode.

That is not how it is supposed to be
If you select in the advanced options "Enable SSH" -> "Allow Public key authentication only" it does is supposed to disallow password logins, as is indeed best practice.

What operating system are you provisioning?
Cannot reproduce the issue with RPI OS.

```
$ ssh pi64.local
Linux pi64 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri Jun  3 09:25:36 2022 from 192.168.178.146
max@pi64:~ $ 
logout
Connection to pi64.local closed.
$ ssh -o PreferredAuthentications=password pi64.local
max@pi64.local: Permission denied (publickey).
```

r-colvin, 2023-03-09:
hi i am using the latest Raspberry Pi Imager and having this issue (need)

in the previous versions of imager, you had to have either ssh key or username/password. in the latest you can do both, but if you do you *must* set a password (unless i am doing something wrong, which is a strong possibility!)

what i would like to do is, im imager use passwordless loging with ssh keys (as all my Pi's are headless and remote); but i want to also change the default username - currently if i want to change the username I need to also set a password

maxnet, 2023-03-09:
>in the previous versions of imager, you had to have either ssh key or username/password. in the latest you can do 
>both, but if you do you must set a password (unless i am doing something wrong, which is a strong possibility!)

Recent RPI OS versions kinda demand that you set a password in all cases.
You can agree or disagree with that, but your system will not work properly until you set one, the nag screen preventing other things from starting.

If you have set a proper SSH public key, the password is not used when logging in through SSH though.
Only for local logins.

maxnet, 2023-03-10:
>so I would like to keep this setup, but be able to set the default user to other than pi, and avoid having the rpi-first-boot-wizard

I understand what you want.
Not setting a password and just SSH username + key + a "do not want the first run wizard" checkbox is something we had before...

But is no longer supported in newer Pi OS releases.
You would need to be able the Pi OS folks first that was a mistake, and the old behavior was better, before this can be changed in Imager again...

r-colvin, 2023-03-10:
@maxnet  yep... so my solution is to create a new user after the image; it is what is 👍

maxnet, 2023-03-10:
>yep... so my solution is to create a new user after the image; it is what is

???
Just set a password in Imager in addition to the SSH key.

r-colvin, 2023-03-10:
oh god... it must be friday afternoon and time to knock off work... you're right of course!

Buzzmovies, 2023-05-02:
Is it just me or I was unable to set a password in Imager for user "pi" or "root".

chrullrich, 2023-07-16:
Imager only disables PasswordAuthentication in sshd_config, it does not disable KbdInteractiveAuthentication (formerly known as ChallengeResponseAuthentication). That option defaults to enabled in stock OpenSSH, no idea about RPiOS.

closed 2024-09-19 by tdewey-rpi