raspberrypi/rpi-imager #445: Windows 11 "advanced" settings do not configure properly

platypushunter, 2022-05-26:
I was trying to use a raspberry pi zero, and put Raspbian lite on there and per my normal procedure I would use the settings (either by shift+ctrl+x or using the new settings button) and setting the wifi etc. 

Well after several times of this not working (being that it wouldnt connect to wifi) I realized that whenever I saved the settings, the password saved is always some random string. In fact both passwords (the one for the pi account and the wifi password) would be some  random strings.

Furthermore the password in the firstrun.sh is the same exact way.

lurch, 2022-05-27:
It's not "some random string", it's a hashed version of the password. See e.g. #406

platypushunter, 2022-05-27:
@lurch I looked at this previously. This is not the same issue. If I use the settings and put in a wifi password, then save it, and burn the image, in wpa_supplicant.conf it doesn't have the same password. Instead of "WiFiPassword" it will have a random string. It may very well be a hash, but regardless it doesn't work.

Then again if I hit "to always use" instead of "for this session only" and I save the settings, then I burn an image, then not only is the password not saved correctly in the wpa_supplicant (let alone the user account) but when I go back to the settings they are not saved correctly, again they are this "random string" 

I uninstalled and reinstalled the program, and the issue persisted. I got it to work once by changing it to "for this session only" and hit "enable ssh" and set the password as "raspberry" and enabling the wifi, but I still had to open wpa_supplicant and change the wifi password to be my actual password instead of the other nonsense.

maxnet, 2022-06-03:
> @lurch I looked at this previously. This is not the same issue. If I use the settings and put in a wifi password, then save it, and
> burn the image, in wpa_supplicant.conf it doesn't have the same [...]

Yes, that's a hash of the SSID + password.
And should work.

Triple check your input...
And also take in mind that the SSID is CaSe SeNSitiVe as well.
Sometimes people conclude there is nothing wrong with the password, but overlook that the SSID does not have the exact same casing as their access point is using.
If you do correct the SSID, do make sure you re-enter the password as well, as the hash will no longer be correct in that case.

maxnet, 2022-08-10:
>but the OS seems to refuse to pick that up

It doesn't delete the firstrun.sh after first boot, or how did you came to the conclusion that it does not process firstboot at all?
It did put "systemd.run=/boot/firstrun.sh systemd.run_success_action=reboot" in cmdline.txt ?

maxnet, 2022-08-10:
Strange. Shows empty in vscode but does is 151 bytes??

cmdline.txt is the last thing we write, before ejecting the drive.
Did you also try a different SD card reader?
Does it work any better if you uncheck the "Eject media when finished" box, and wait a while before removing the SD card?
Just in case the SD card reader has a bug in which it doesn't write out memory buffers to storage, if told to eject.

maxnet, 2022-08-10:
>Is this supposed to happen? Isn't all empty space should be allocated to rootfs?

That is also done by cmdline.txt (it has init=some-script-that-does-the-resize-on-first-boot.sh)
So yes, if your file system is corrupt to the point that cmdline.txt shows up empty, the ext4 partition does not get resized on first boot either.

The question do is why it becomes corrupt.

maxnet, 2022-08-10:
>But I am wondering, does rpi-imager verify the modified cmdline.txt file?

No.
That is harder to do than you may think.
If we read back the file immediately after writing it, chances are very high Windows (or the SD card reader itself) will "read" the file from its memory cache, instead of the data actually coming from SD card.

Does it work correctly with the problematic SD card reader if you uncheck "eject media when finished" and wait a while before removing SD card?
We do may be able to add an extra second delay before ejecting media if that SD card readers just need a little bit more time to write out data before ejecting.

lurch, 2022-08-10:
If RPi Imager wrote the `cmdline.txt` file _before_ the verification process, then the verification process would always fail because you've changed what data is on the SD card compared to the original image file :wink:  That's why "cmdline.txt is the last thing we write" as @maxnet says (after the verification process has finished).

maxnet, 2022-08-10:
Does it concern a SD card reader that can be easily obtained on for example AliExpress or Amazon, and if so do you have a link to the product page?

I am curious if I would be able to reproduce the problem.

maxnet, 2022-08-18:
[image]

Unfortunately, I am not able to reproduce the issue with that reader.
cmdline.txt was edited and firstrun.sh is there.

So issue may be very timinig specific, or also depending on other factors of your system.
Unfortuneately, kinda hard to debug.

matsonfruit, 2022-09-02:
same issue here. The advanced features don't do anything for me. I'm on Windows 11 using the microsd card reader built-into my thinkpad laptop

maxnet, 2022-09-02:
>The advanced features don't do anything for me.

There is no firstrun.sh being created?
cmdline.txt is not modified?

kayg04, 2023-02-09:
There is no firstrun.sh. Although the `cmdline.txt` mentions it:

```
console=serial0,115200 console=tty1 root=PARTUUID=21e60f8c-02 rootfstype=ext4 fsck.repair=yes rootwait quiet init=/usr/lib/raspberrypi-sys-mods/firstboot systemd.run=/boot/firstrun.sh systemd.run_success_action=reboot systemd.unit=kernel-command-line.target
```

[image]

kayg04, 2023-02-09:
Creating an image with rpi 1.7.2 solves the issue.

[image]

tdewey-rpi, 2024-08-28:
Closing as resolved - as of 1.7.2.

closed 2024-08-28 by tdewey-rpi