raspberrypi/rpi-imager #446: Build error on Debian buster

michelepagot, 2022-05-29:


michelepagot, 2022-05-29:
Error is like 
`CMake Error at CMakeLists.txt:99 (add_executable):
  Target "rpi-imager" links to target "LibLZMA::LibLZMA" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?`
  
  
  build instructions
  `$ git clone --depth 1 https://github.com/raspberrypi/rpi-imager
   $  cd rpi-imager
   $  debuild -uc -us`

System details
`$ uname -a
Linux jervis02 4.19.0-20-amd64 #1 SMP Debian 4.19.235-1 (2022-03-17) x86_64 GNU/Linux`

And
`$ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"`

All  required prerequisites are installed:
`sudo apt install --no-install-recommends build-essential devscripts debhelper cmake git libarchive-dev libcurl4-openssl-dev qtbase5-dev qtbase5-dev-tools qtdeclarative5-dev libqt5svg5-dev qttools5-dev libssl-dev qml-module-qtquick2 qml-module-qtquick-controls2 qml-module-qtquick-layouts qml-module-qtquick-templates2 qml-module-qtquick-window2 qml-module-qtgraphicaleffects
Reading package lists... Done
Building dependency tree
Reading state information... Done
libcurl4-openssl-dev is already the newest version (7.64.0-4+deb10u2).
git is already the newest version (1:2.20.1-2+deb10u3).
libarchive-dev is already the newest version (3.3.3-4+deb10u1).
libssl-dev is already the newest version (1.1.1n-0+deb10u2).
build-essential is already the newest version (12.6).
cmake is already the newest version (3.13.4-1).
debhelper is already the newest version (12.1.1).
devscripts is already the newest version (2.19.5+deb10u1).
qtbase5-dev is already the newest version (5.11.3+dfsg1-1+deb10u4).
qtbase5-dev-tools is already the newest version (5.11.3+dfsg1-1+deb10u4).
qml-module-qtquick-layouts is already the newest version (5.11.3-4).
qml-module-qtquick-window2 is already the newest version (5.11.3-4).
qml-module-qtquick2 is already the newest version (5.11.3-4).
qtdeclarative5-dev is already the newest version (5.11.3-4).
qml-module-qtgraphicaleffects is already the newest version (5.11.3-2).
qml-module-qtquick-controls2 is already the newest version (5.11.3+dfsg-2).
qml-module-qtquick-templates2 is already the newest version (5.11.3+dfsg-2).
libqt5svg5-dev is already the newest version (5.11.3-2).
qttools5-dev is already the newest version (5.11.3-4).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
`

And liblzma-dev too:
`$ sudo apt-get install liblzma-dev liblzma5
Reading package lists... Done
Building dependency tree
Reading state information... Done
liblzma-dev is already the newest version (5.2.4-1+deb10u1).
liblzma5 is already the newest version (5.2.4-1+deb10u1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.`

michelepagot, 2022-05-29:
This one works fine:
```
$ mkdir BUILD
$ cd BUILD
$ cmake -GNinja ../src
$ ninja
```

maxnet, 2022-06-03:
```
dh_auto_configure: cd obj-x86_64-linux-gnu && cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=None -DCMAKE_INSTALL_SYSCONFDIR=/etc -DCMAKE_INSTALL_LOCALSTATEDIR=/var -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=ON "-GUnix Makefiles" -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_INSTALL_LIBDIR=lib/x86_64-linux-gnu ../src returned exit code 1
```

What are the lines directly above this.

lurch, 2022-06-04:
Strange. I just tried running the Debian build instructions in the README inside fresh Docker images for both Buster and Bullseye (on my Ubuntu x64 laptop), and they both ran successfully (with warnings about ZLIB and LibLZMA not being available); but both builds ran to completion without generating any errors.

Sebourg, 2022-09-28:
> ```
> dh_auto_configure: cd obj-x86_64-linux-gnu && cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=None -DCMAKE_INSTALL_SYSCONFDIR=/etc -DCMAKE_INSTALL_LOCALSTATEDIR=/var [...]

Same issue here, on:
```
$ uname -a
Linux duh 4.19.0-21-686-pae #1 SMP Debian 4.19.249-2 (2022-06-30) i686 GNU/Linux

$ cmake --version
cmake version 3.13.4

CMake suite maintained and supported by Kitware (kitware.com/cmake).

```

Here are the lines just above, if it's any help:
```
$ debuild -uc -us
 dpkg-buildpackage -us -uc -ui
dpkg-buildpackage: info: source package rpi-imager
dpkg-buildpackage: info: source version 1.7.3
dpkg-buildpackage: info: source distribution unstable
dpkg-buildpackage: info: source changed by Floris Bos <bos@je-eigen-domein.nl>

...

make[2]: Leaving directory '/home/duh/Projects/rpi-imager/obj-i686-linux-gnu/CMakeFiles/CMakeTmp'
make[1]: Leaving directory '/home/duh/Projects/rpi-imager/obj-i686-linux-gnu/CMakeFiles/CMakeTmp'

Source file was:

    #include <atomic>
    #include <stdint.h>
    int main() {
        std::atomic<int64_t> x;
        x = 1;
        return (int) x;
    }
dh_auto_configure: cd obj-i686-linux-gnu && cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=None -DCMAKE_INSTALL_SYSCONFDIR=/etc -DCMAKE_INSTALL_LOCALSTATEDIR=/var -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=ON "-GUnix Makefiles" -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_INSTALL_LIBDIR=lib/i386-linux-gnu ../src returned exit code 1
make: *** [debian/rules:7: build] Error 2
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2
debuild: fatal error at line 1182:
dpkg-buildpackage -us -uc -ui failed
```
Running cmake alone
```
$ cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=None -DCMAKE_INSTALL_SYSCONFDIR=/etc -DCMAKE_INSTALL_LOCALSTATEDIR=/var -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=ON "-GUnix Makefiles" -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_INSTALL_LIBDIR=lib/i386-linux-gnu ../src
udisks2 support enabled
Enabled LTO
{
   "blockdevices": [
      {"name":"sda", "maj:min":"8:0", "rm":false, "size":"59.6G", "ro":false, "type":"disk", "mountpoint":null,
         "children": [
            {"name":"sda1", "maj:min":"8:1", "rm":false, "size":"100M", "ro":false, "type":"part", "mountpoint":null},
            {"name":"sda2", "maj:min":"8:2", "rm":false, "size":"49.1G", "ro":false, "type":"part", "mountpoint":null},
            {"name":"sda3", "maj:min":"8:3", "rm":false, "size":"450M", "ro":false, "type":"part", "mountpoint":null},
            {"name":"sda4", "maj:min":"8:4", "rm":false, "size":"1K", "ro":false, "type":"part", "mountpoint":null},
            {"name":"sda5", "maj:min":"8:5", "rm":false, "size":"8G", "ro":false, "type":"part", "mountpoint":"/"},
            {"name":"sda6", "maj:min":"8:6", "rm":false, "size":"2G", "ro":false, "type":"part", "mountpoint":"[SWAP]"}
         ]
      },
      {"name":"sdb", "maj:min":"8:16", "rm":true, "size":"7.4G", "ro":false, "type":"disk", "mountpoint":null}
   ]
}
-- Configuring done
CMake Error at CMakeLists.txt:109 (add_executable):
  Target "rpi-imager" links to target "LibLZMA::LibLZMA" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at CMakeLists.txt:109 (add_executable):
  Target "rpi-imager" links to target "GnuTLS::GnuTLS" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

-- Generating done
-- Build files have been written to: /home/duh/Projects/rpi-imager/obj-i686-linux-gnu
```

maxnet, 2022-09-29:
>cmake version 3.13.4

The imported target GnuTLS::GnuTLS that we use, is only available in CMake 3.16 (and LibLZMA::libLZMA is only available in 3.14), so that things do not work that well with CMake 3.13 is to be expected.

It may be possible to rewrite the cmake file to use the legacy CMake constructs, in which you have to add GNUTLS_LIBRARIES to the libraries, GNUTLS_INCLUDE_DIR to the includes, GNUTLS_DEFINITIONS to the cmake defines, etc.
Instead of the modern way of just refering to GnuTLS::GnuTLS, and have cmake take care of all those things automagically.
But not sure if that's worth the effort, given that Buster is pretty soon EOL.

Sebourg, 2022-09-29:
Fair enough. 
I did give up, as I ran into an other issue after building a custom version of cmake.
Qt version on Buster is 5.11 and it looks like rpi-imager 1.7.3 use an Qt namespace (QPasswordDigestor) introduce in Qt 5.12.

maxnet, 2022-09-29:
>Qt version on Buster is 5.11 and it looks like rpi-imager 1.7.3 use a Qt namespace (QPasswordDigestor) introduce in Qt 5.12.

You would need to reverse the following commit if you want to use an old Qt version:

https://github.com/raspberrypi/rpi-imager/commit/cb415a6a026ff91da1f5e93933e10c940641f402

Then you do can use an older Qt version, but do will have a dependency on OpenSSL for calculating the digest instead.
Dependencies on OpenSSL are not so great, because OpenSSL's ABI is not binary safe between OpenSSL versions.
Meaning that if you would upgrade your Debian to a later version at some point, that has a newer version of OpenSSL, you would need to recompile Imager again...

Sebourg, 2022-09-30:
@maxnet Thanks for those suggestions, I ended up successfully building version 1.7.2 instead of 1.7.3.

derrelldurrett, 2024-01-06:
I'm not sure, but I think I'm seeing what is functionally the same as this error: https://github.com/raspberrypi/rpi-imager/issues/446#issuecomment-1262639709

I didn't have to build a custom cmake (or I don't think I did).

```
$ cmake --version
cmake version 3.13.4

CMake suite maintained and supported by Kitware (kitware.com/cmake).
$ uname -a
Linux 4.19.0-25-amd64 #1 SMP Debian 4.19.289-2 (2023-08-08) x86_64 GNU/Linux
$ cd obj-x86_64-linux-gnu && cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=None -DCMAKE_INSTALL_SYSCONFDIR=/etc -DCMAKE_INSTALL_LOCALSTATEDIR=/var -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=ON "-GUnix Makefiles" -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_INSTALL_LIBDIR=lib/x86_64-linux-gnu ../src
CMake Error at CMakeLists.txt:109 (message):
  Missing suitable Qt library (must be at least version 5.14)

-- Configuring incomplete, errors occurred!
See also "/home/derrell/INSTALLS/rpi-imager/obj-x86_64-linux-gnu/CMakeFiles/CMakeOutput.log".
```

(The last is after `debuild -uc -us` fails and displays the `cmake` command that fails:

```
$ debuild -uc -us
... <a lot of stuff> ...
TestEndianess.c:
/* A 16 bit integer is required. */
typedef unsigned short cmakeint16;

/* On a little endian machine, these 16bit ints will give "THIS IS LITTLE ENDIAN."
   On a big endian machine the characters will be exchanged pairwise. */
const cmakeint16 info_little[] =  {0x4854, 0x5349, 0x4920, 0x2053, 0x494c, 0x5454, 0x454c, 0x4520, 0x444e, 0x4149, 0x2e4e, 0x0000};

/* on a big endian machine, these 16bit ints will give "THIS IS BIG ENDIAN."
   On a little endian machine the characters will be exchanged pairwise. */
const cmakeint16 info_big[] =     {0x5448, 0x4953, 0x2049, 0x5320, 0x4249, 0x4720, 0x454e, 0x4449, 0x414e, 0x2e2e, 0x0000};

#ifdef __CLASSIC_C__
int main(argc, argv) int argc; char *argv[];
#else
int main(int argc, char *argv[])
#endif
{
  int require = 0;
  require += info_little[argc];
  require += info_big[argc];
  (void)argv;
  return require;
}

dh_auto_configure: cd obj-x86_64-linux-gnu && cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=None -DCMAKE_INSTALL_SYSCONFDIR=/etc -DCMAKE_INSTALL_LOCALSTATEDIR=/var -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=ON "-GUnix Makefiles" -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_INSTALL_LIBDIR=lib/x86_64-linux-gnu ../src returned exit code 1
make: *** [debian/rules:7: build] Error 2
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2
debuild: fatal error at line 1182:
dpkg-buildpackage -us -uc -ui failed
```

Repeating it shows the error regarding the Qt library version required.)

But I have libQt5 version 5.15 in `/usr/lib/x86_64-linux-gnu/`.

My assumption was that I could modify the make files, but I haven't successfully identified how to do that [by reading cmake documentation](https://cmake.org/cmake/help/latest/command/link_directories.html):
```
$ find . -type f -exec grep -H link_directories {} \;
./src/dependencies/cmcurl/CMakeLists.txt:    link_directories(${GSS_LINK_DIRECTORIES})
```
I've tried adding the line `link_directories(/usr/lib/x86_64-linux-gnu)` to `src/CMakeLists.txt` before the `find_package(Qt5 ...)` line.

If additional information would be helpful, please let me know. CMake isn't my bailiwick, so I don't know what else to try.

derrelldurrett, 2024-01-25:
I finally played around with this long enough that I've gotten to exactly [where @Sebourg was](https://github.com/raspberrypi/rpi-imager/issues/446#issuecomment-1261575598).

To do that, I've done this:
```
 set(Qt5_DIR "../../qt-everywhere-src-5.15.2/qtbase/lib/cmake/Qt5")
link_directories(BEFORE "/usr/lib/x86_64-linux-gnu" "../../qt-everywhere-src-5.15.2")
```

Moving this line to before where the `find_package` call for LibLZMA and GnuTLS didn't resolve this issue.

tdewey-rpi, 2024-09-18:
Closing as stale - no activity since January, all existing comments are not likely to provide useful context any new issues.

closed 2024-09-18 by tdewey-rpi