raspberrypi/rpi-imager #447: Faulty OctoPi 1.8.1 Instance - RPI-Imager v1.7.2

Twarner491, 2022-06-01:
Although flashing works as desired, OctoPi instance 1.8.1 found under ```OS < Other specific-purpose OS < 3D printing < OctoPi``` in RPI-Imager v 1.7.2 seems to be a faulty instance. Upon initial bootup, octoprint instances running OctoPi from the RPI Imager yield the _"The OctoPrint server is currently not running page"_ error page. I've proved the effectiveness of OctoPi v1.8.1 via flashing the OS to a RPI-4B with the image provided on [Octoprint's Downloads Page](https://octoprint.org/download/) and balentaEtcher - yet have achieved constant errors from the image provided by RPI-Imager across 7 different RPI-4b and 7 different Prusa Mini+. The instance is still accessible via SSH, yet the GUI will not properly render. All 7 printers, RPI-4bs & SD cards are proved effective with other methods of flashing the same Octoprint 1.8.1 instance, thus a faulty OctoPi 1.8.1 image in RPI-Imager v1.7.2 can be assumed.

Please follow up if any additional data is needed.

Thanks!

maxnet, 2022-06-03:
>Although flashing works as desired

Then it does not look like an Imager problem to me, but something better reported to OctoPi.

ping @guysoft

guysoft, 2022-06-03:
What version of OctoPI? OctoPi has no version 1.8.1. 1.0.0 has not been released yet.

Twarner491, 2022-06-03:
My mistake, Octoprint V1.8.1, OctoPi v0.18.0

guysoft, 2022-06-04:
i am not really sure how to debug this, since most of your input is stuff you did with OctoPrint, which results in "does not work" and long load page times, which is Octoprint-related and not OctoPi. I suggest you post the octoprint.log to the OctoPrint issue tracker:
https://github.com/foosel/OctoPrint/issues/

lurch, 2022-06-04:
I just did a bit of digital-sleuthing, and https://downloads.raspberrypi.org/os_list_imagingutility_v3.json (which is the file that RPi Imager uses to download OS images) points to https://octopi.octoprint.org/rpi-imager.json , and the image filename mentioned in there ( https://github.com/OctoPrint/OctoPi-UpToDate/releases/download/0.18.0-1.8.1-20220524101555/octopi-0.18.0-1.8.1-20220524101555.zip ) is identical to the one mentioned on https://octoprint.org/download/ :confused:

Twarner491, 2022-06-11:
@lurch, thanks for the additional sleuthing, I'll attempt another couple of instances via RPI over the weekend & may follow @guysoft's advice to open an OctoPrint-specific issue. Thanks again.

tdewey-rpi, 2023-11-17:
Closing as not an Imager issue.

closed 2023-11-17 by tdewey-rpi