raspberrypi/rpi-imager #451: "Advanced options" inaccessible after launching Imager by file association

kawadakk, 2022-06-09:
We are building a custom OS image that is based on [pi-gen](https://github.com/RPi-Distro/pi-gen). During testing, we found that the "advanced options" button, which is essential for our use case, does not show up after launching Imager by double-clicking an image file. It does though when you launch Imager manually and then choose the image file.

lurch, 2022-06-09:
Have you got both an old and new version of RPi Imager installed? Maybe launching Imager manually launches the new version, but double-clicking on an image file launches the old version?

kawadakk, 2022-06-09:
No, the title bar says v1.7.2 in both cases.

kawadakk, 2022-06-09:
I just tested and it seems this issue can be reproduced with [an official image](https://www.raspberrypi.com/software/operating-systems/) (e.g., https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2022-04-07/2022-04-04-raspios-bullseye-armhf-lite.img.xz), too.

lurch, 2022-06-09:
Oh yes, there's an `init_format` setting in the http://downloads.raspberrypi.org/os_list_imagingutility_v3.json file read by RPi Imager to determine if the advanced options should be shown, see https://github.com/raspberrypi/rpi-imager/blob/qml/debian/changelog#L20

PR #642 (2023-10-06): Show options when opening image by file association
If you load a custom image by running `rpi-imager my_image.img.xz` then the Advanced Options button is not shown.  This is triggered if you set a file association to load rpi-imager and then double-click an image file.

The fix is simply to set the visibility on this button on start-up (custom images set on the command-line are loaded before this).

Fixes #451
commit dbf97577 (2023-10-06): Show options when opening image by file association

If you load a custom image by running `rpi-imager my_image.img.xz` then
the Advanced Options button is not shown.  This is triggered if you set
a file association to load rpi-imager and then double-click an image
file.

The fix is simply to set the visibility on this button on start-up
(custom images set on the command-line are loaded before this).

Fixes #451
closed 2023-10-06 by maxnet