raspberrypi/rpi-imager #452: Feature Request: Multiple WiFi Networks

makermelissa, 2022-06-13:
I know the Raspberry Pi OS supports multiple WiFi networks by adding more entries to the wpa_supplicant.conf file. It would be cool if multiple networks could be set in the imager and then they are added at imaging time.

lurch, 2022-06-14:
I'm only guessing (I don't work on this project myself, but I try to be helpful in the issues), but this sounds like such a niche use-case that it'll never get added to RPi Imager? :man_shrugging:

tdewey-rpi, 2024-02-07:
Rejecting feature request.

While I would _love_ to have this feature for myself, I don't believe it's going to be common-enough of a use case for the typical Raspberry Pi Imager user that we'll explore it in the short term. Never say never, though.

Thanks for the submission, @makermelissa, and I'm happy to review and provide feedback on a PR if you submit one.

closed 2024-02-07 by tdewey-rpi

makermelissa, 2024-02-07:
Fair enough. I had kind of forgot about this.