raspberrypi/rpi-imager #454: BUG: rpi-imager 1.7.2 gets write error on macOS 13 beta (22A5266r)

arag0re, 2022-06-22:
# BUG: imager 1.7.2 gets write error on macOS 13 beta (22A5266r)

As you can see in the screenshot linked below, im getting a write error with every OS-image I select and also with every sd-card reader I used. Im a developer myself and i would love to help out if i can.
The System I use is the [macbook 18,3](https://support.apple.com/en-us/HT204384). I tried the internal sd-card reader and multiple external usb sd-card reader, all of them got this error after entering sudo password.

[image]

maxnet, 2022-06-22:
>macOS 13 beta

Haven't tested the beta yet.

Did notice the release notes do mention new security restrictions (that can behave different depending on the Mac you have).
But not sure whether or not that may be related to the problem you are seeing.

https://developer.apple.com/documentation/macos-release-notes/macos-13-release-notes

```
Accessory Security
New Features
On portable Mac computers with Apple silicon, new USB and Thunderbolt accessories require user approval before the accessory can communicate with macOS for connections wired directly to the USB-C port. This doesn’t apply to power adapters, standalone displays, or connections to an approved hub. Devices can still charge if you choose Don’t Allow.

You can change the security configuration in System Settings > Security and Privacy > Security. The initial configuration is Ask for new accessories. Configuring an accessibility Switch Control sets the policy to always allow accessory use. Approved devices can connect to a locked Mac for up to three days.

Accessories attached during software update from prior versions of macOS are allowed automatically. New accessories attached prior to rebooting the Mac might enumerate and function, but won’t be remembered until connected to an unlocked Mac and explicitly approved. (43338666)
```

arag0re, 2022-06-22:
> 

I dont think thats the issue I have, cause i approved the usb card reader in the dialog macOS shows you when connecting a new usb-device. Also I erased it with diskutil and it worked as expected.

arag0re, 2022-06-23:
Error is resolved in beta 2 of macOS 13
Tho I have to restart the write process several times for it to work. same issue I had in macOS12.3.1

maxnet, 2022-06-23:
>Tho I have to restart the write process several times for it to work. same issue I had in macOS12.3.1

Is that also the case when using the external USB SD card reader, or just with the internal one?

(As others have reported problems with the internal one being unresponsive before: https://github.com/raspberrypi/rpi-imager/issues/270#issuecomment-999467151 )

arag0re, 2022-06-24:
> > Tho I have to restart the write process several times for it to work. same issue I had in macOS12.3.1
> 
> Is that also the case when using the external USB SD card reader, or just with the internal one?
> 
> (As others have reported problems with the internal one being unresponsive before: [#270 (comment)](https://github.com/raspberrypi/rpi-imager/issues/270#issuecomment-999467151) )

It's unresponsive with internal and external readers for me. Tho u have to keep in mind that I only had an sandisk usb-c reader on hand.