raspberrypi/rpi-imager #457: Pull wifi password automatically

cfpwastaken, 2022-07-07:
I always use the advanced options to make my pi connect to my wifi, but i always have to get the password myself and paste it in.
Ubuntu can just tell you the password from its settings, windows can tell you the password using a command.
So why cant the imager autofill the password if it find one on the system?

maxnet, 2022-07-07:
Main problem is that on Linux that requires root privileges, and our program runs as normal user.

cfpwastaken, 2022-07-07:
1. You could just make a [ PULL PASSWORD ] button that then asks for root, right?
2. Hmm the settings never ask me for any root password, maybe there is some systemd service running serving that password? Like shutdown/reboot

maxnet, 2022-07-07:
>You could just make a [ PULL PASSWORD ] button that then asks for root, right?

Normally needs to be done before starting the program.
Can ask it for specific actions later if offered through some DBus service. (Like udisks2 which we ask to open the raw disk device for us).

>Hmm the settings never ask me for any root password

Maybe that is a feature offered by the network manager used by your Ubuntu flavor.
But that's not standard Linux behavior, so hard to rely on.
wpa_supplicant itself (that handles the wifi authentication) only allows password to be set through DBus, not read.

commit ef4abc66 (2022-07-31): Linux distros that use NetworkManager: prefill WLAN PSK

Only supported when NetworkManager is used on the Linux distro
as wpa_supplicant itself does not support reading PSKs through
DBus, only setting them.

Closes #457
closed 2022-07-31 by maxnet