raspberrypi/rpi-imager #459: Add additional netplan keys for wifi  (ubuntu)

mmattel, 2022-07-08:
When installing an OS that uses netplan like U22.04 server via the imager, you get asked about the WIFI data which is then written to the netplan config file. Two suggestions to add:

* First add the `optional: true` key which deblocks booting when the access point is down.
* Add a new field in the imager to ask if the WIFI is hidden and add in case the `hidden: true` option in the netplan config. Else you get a hard time finding why it does not connect.

See a working example:
```
     wifis:
        wlan0:
            optional: true
            access-points:
                ABCD:
                  password: "*****"
                  hidden: true
...
```

maxnet, 2022-07-08:
>First add the optional: true key which deblocks booting when the access point is down.

Will need testing to see if the code to install the avahi .deb package if the user set a .local hostname still works with that.

>Add a new field in the imager to ask if the WIFI is hidden

We already have a "hidden SSID" checkbox.

maxnet, 2022-08-11:
Actually we already have "optional: true" as well.
https://github.com/raspberrypi/rpi-imager/blob/4961990ce00892325b08a10d91f5eba9b3711e7c/src/OptionsPopup.qml#L738

Closing this.

closed 2022-08-11 by maxnet