raspberrypi/rpi-imager #460: Use without GUI from cli-script

hiqsol, 2022-07-11:
I want to write images from script without GUI passing all options as cli-arguments or config file.
Is there a way to use the app in such way?
Thanks!

maxnet, 2022-07-11:
For what use-case exactly?

We do have a CLI mode, but it requires that you know the exact device name you are writing to.
Making it not so suitable if you are planning on distributing that script to others, if you have no idea what storage they are using.

[code]
rpi-imager --cli [--disable-verify] [--sha256 <expected hash>] [--debug] [--quiet] <image file to write> <destination drive device>
[/code]

Also note that if you are using Pi inside some kind of product, and want to provision the storage of that, other options such as cmprovision also exist: https://github.com/raspberrypi/cmprovision
Although written for provisioning Compute Modules it can also provision Pi 4, if you insert a SD card in them first that enables PXE network boot in EEPROM settings.

lurch, 2022-07-11:
Little bit more info about CLI-mode in #221

maxnet, 2022-07-11:
>AFAICS rpi-imager --cli does not support advanced options (ssh, wifi). Does it? 

No.

>And I did not yet understood if cmprovision does it

There are no standard checkboxes there for things like wifi there, as cmprovision is typically used to mass-program units by manufacturers, and those do not know which wifi password the end-user will have.

Instead it does have support for running configurable post-installation shell scripts on the unit being provisioned.
For things like resizing the root file system, so that this no longer needs to be done on first boot, or whatever else you need to have done.

If you were using it to provision Ubuntu you do could also let it create a cloud-init "user-data" file on the FAT partition with your desired settings.
But it does require some effort on your part to draft that file, and it is not a ready made "enable wifi" checkbox.

For illustrative purposes:

cmprovision scripts screen where you can edit scripts or add your own:

[image]

Project screen where you can select which image should be written and what scripts you created earlier you want to apply:

[image]

closed 2022-09-08 by maxnet

P33M, 2024-09-03:
Hi, the CLI feature is slightly broken as of 1.8.5.

If `rpi-imager` is invoked on a headless terminal, I get:
```
pi@raspberrypi:~$ rpi-imager
qt.qpa.xcb: could not connect to display
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, xcb.

Aborted
```
Which prevents me from discovering command-line arguments. After learning about it elsewhere, calling with `--cli` works.