raspberrypi/rpi-imager #462: [OS Request] Bareboat Necessities OS

mgrouch, 2022-07-15:
Hello,

I'm maintainer of Bareboat Necessities OS. Can you, please, add it under "Other specific-purpose OS"?

https://raw.githubusercontent.com/bareboat-necessities/my-bareboat/master/rpi-imager/rpi-imager-bbn.json

```
// Snippet to include Bareboat Necessities OS Image in Raspberry Pi Imager under
// "Other specific-purpose OS"

{
  "name": "Bareboat Necessities Open-Source Marine Linux OS",
  "description": "Official Raspberry Pi4/400 Image for Bareboat Necessities Open-Source Marine Linux OS.",
  "icon": "https://raw.githubusercontent.com/bareboat-necessities/my-bareboat/master/bareboat-necessities-logo.svg",
  "subitems_url": "https://raw.githubusercontent.com/bareboat-necessities/my-bareboat/master/rpi-imager/rpi-imager-bbn.json"
},

```

Thanks,
--MG

ghollingworth, 2022-07-15:
Is that size correct?  11GBytes?

mgrouch, 2022-07-15:
Yes

2.8 in xz. I test all releases with rpi-imager. 
11 in .img

tdewey-rpi, 2023-11-17:
Please fill out the form to request an addition to the OS list at https://forms.gle/cbsU27aLLVaxUXbw8.

closed 2023-11-17 by tdewey-rpi