raspberrypi/rpi-imager #465: No Windows Start Menu folder/shortcuts

ollie-campbell, 2022-08-03:
After installation I was expecting the Imager to have a Start Menu folder/shortcuts. It seems it doesn't.

The app is fully installed and works, just no Start menu shortcuts.

I'm currently using Windows 11, but I think this was also happening towards the end of my Windows 10 usage days, although I can't be 100% sure!

maxnet, 2022-08-03:
They are put in the folder "Raspberry Pi"

ollie-campbell, 2022-08-03:
> 

Not for me I'm afraid....

maxnet, 2022-08-03:
If you press [windows key+R], and go to "%appdata%\Microsoft\Windows\Start Menu\Programs", there is no "Rasperry Pi" folder there either?
You did install Imager as the user you are currently logged in as? As it is not installed for all Windows users.

ollie-campbell, 2022-08-03:
> If you press [windows key+R], and go to "%appdata%\Microsoft\Windows\Start Menu\Programs", there is no "Rasperry Pi" folder there either? You did install Imager as the user you are currently logged in as? As it is not installed for all Windows users.

The folder exists but it's empty....

maxnet, 2023-03-03:
Sorry, for the late reply, but are you using a virus scanner, and if so, which one?

It seems 3 scanners have started to misidentify the "start menu" .lnk file NSIS creates during installation as a threat...

https://www.virustotal.com/gui/file/00148fc8163870a526696fbe7c24b808c410de154d56c586de9c786ceddad779/detection

```
Raspberry Pi Imager.lnk

Security vendors' analysis

Fortinet  LNK/Runner.BQ!tr
Google    Detected
Ikarus    Trojan.Lnk.Runner
```

ollie-campbell, 2023-03-06:
> Sorry, for the late reply, but are you using a virus scanner, and if so, which one?
>
> It seems 3 scanners have started to misidentify the "start menu" .lnk file NSIS creates during installation as a [...]

I'm using Sophos, so not one that is detected via that link. Also, from memory, nothing flagged on the Sophos portal post installation. But then again I've not tried again since this.

maxnet, 2023-03-06:
>I'm using Sophos, so not one that is detected via that link.

Thanks for the feedback.
Then I do not know what may have caused this in your case. :-(

Meanwhile, Ikarus fixed their detection.

```
Dear Ladies and Gentlemen,
Sehr geehrte Damen und Herren,

Many thanks for the delivered file.

***** false-positive *****

The false positive was removed and should not occur any more after our next database update.

This is an automatic generated e-mail, please do not reply.

Vielen Dank für die von Ihnen eingesandte Datei.

***** Fehlalarm *****

Bei dieser Datei handelt es sich um einen Fehlalarm. Dieser wurde ausgebaut und wird mit dem nächsten Update nicht mehr auftreten.

Dies ist eine automatisch generierte Nachricht, bitte antworten Sie nicht auf diese Nachricht.

Kind Regards,
Mit freundlichen Grüßen

----------------------------------
IKARUS Security Software GmbH
```

Also wrote Fortinet but no response so far.
Not sure how to reach Google. Virustotal only pointing to www.google.com as vendor contact does not help...

tdewey-rpi, 2025-12-12:
This is now a stale report, and we even use a different installer framework (Inno, vs nsis).