raspberrypi/rpi-imager #466: Apple M1 macOS - corrupting SD cards

miklosakos, 2022-08-05:
Creating an SD card with rpi-imager results in an error message stating "error writing image" and the end result is an SD card with no partition table what so ever. Running dd from the terminal onto the same SD card will make a proper image.
Formatting the SD card works. I'm using a 2021 MacBook Pro (with an 8 core M1 Pro CPU and macOS 12.5) and it's built-in SD card reader with a 16GB SanDisk card.
During the same fail session I came across the following: "Error reading from storage. SD card may be broken"

maxnet, 2022-08-05:
You may want to try if it works any better if you use an external USB SD card reader.

There has been another report about the internal reader (that is only present on the higher-end Macbook Pro models) sometimes not behaving properly: https://github.com/raspberrypi/rpi-imager/issues/270#issuecomment-999467151

Other user is mentioning it stops responding for a while after FAT32 format. Which is not done in our code, but outsourced to Apple's own diskutil. If it turns out there are deeper issues like this with this drive, this is unlilely something we can fix.

miklosakos, 2022-08-05:
When I did "Erase" on the card through Raspberry Pi Imager I didn't notice any hangs with diskutility or the system in general with the SD card present, in fact I could interact with it just fine through the GUI tools, I'll take another look at the issue to see if I can repro the comment you mentioned.

miklosakos, 2022-08-05:
I did a quick test with the Kioxia SD card, erase went through the process just fine and diskutil list did not freeze.
However when the imager tries acquiring stuff from the web the output ends with Cannot allocate memory.
Attached both logs. In this test I tried flashing PINN.
After the failed flash process the SD card has no partitions according to diskutil list.
[rpi-imager_erase.log](https://github.com/raspberrypi/rpi-imager/files/9274014/rpi-imager_erase.log)
[rpi-imager_failed-write.log](https://github.com/raspberrypi/rpi-imager/files/9274015/rpi-imager_failed-write.log)

miklosakos, 2022-08-06:
So even though the process completed successfully, the SD card still has no partitions on it.
[rpi-imager_success.log](https://github.com/raspberrypi/rpi-imager/files/9274103/rpi-imager_success.log)

miklosakos, 2022-08-06:
I went through some older issues and found #235, the universal binary posted [here](https://github.com/raspberrypi/rpi-imager/issues/235#issuecomment-999791765) did in fact work and managed to produce an SD card with partitions on it.
So whatever is going wrong with the latest version can be caused by Rosetta 2 or whatever magic happens in the background.

maxnet, 2022-08-07:
>So whatever is going wrong with the latest version can be caused by Rosetta 2 or whatever magic happens in the background.

Could be, but I would still be interested if the issue also occurs if you are using an external USB SD card reader instead.
As I know it does work correctly under Rosetta for others that have entry level Macbooks (that do not have the internal reader, so I am sure they are using an external one).

>However when the imager tries acquiring stuff from the web the output ends with Cannot allocate memory.

Note that this error message comes from the kernel after we call the write() syscall.
If the kernel runs out of some fixed size memory pool, there is usually something wrong on the kernel side, rather than with the application.
Could be something with the driver for the internal reader, but also anything else that operates in kernel space (drivers for other hardware you may have attached, some anti-virus software that installs kernel extensions, etc.)
If you get such error, make sure you reboot, before conducting further tests.

miklosakos, 2022-08-08:
The weird thing is that only the Intel binary has this issue, the Universal binary works fine. The only external sd card reader I have other than a pinephone is a multi-decade old one that refuses to work with PCs at all. I hooked up a known to work USB Flash Drive and got the same issue of successful write and no usable partition. The same flash drive shows up in Windows with a single, unformatted partition. The same flash drive can be imaged with `dd` and the Windows version of the imager just fine.
`kextstat` output:
```
root@pocketable-nuclear-power-plant Downloads # kextstat
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
Index Refs Address            Size       Wired      Name (Version) UUID <Linked Against>
    1  198 0                  0          0          com.apple.kpi.bsd (21.6.0) 43BBF43C-8008-3830-8A00-C8706889EEA1 <>
    2   11 0                  0          0          com.apple.kpi.dsep (21.6.0) 43BBF43C-8008-3830-8A00-C8706889EEA1 <>
    3  231 0                  0          0          com.apple.kpi.iokit (21.6.0) 43BBF43C-8008-3830-8A00-C8706889EEA1 <>
    4  233 0                  0          0          com.apple.kpi.libkern (21.6.0) 43BBF43C-8008-3830-8A00-C8706889EEA1 <>
    5  218 0                  0          0          com.apple.kpi.mach (21.6.0) 43BBF43C-8008-3830-8A00-C8706889EEA1 <>
    6  142 0                  0          0          com.apple.kpi.private (21.6.0) 43BBF43C-8008-3830-8A00-C8706889EEA1 <>
    7  152 0                  0          0          com.apple.kpi.unsupported (21.6.0) 43BBF43C-8008-3830-8A00-C8706889EEA1 <>
    8   19 0xfffffe0007773e50 0x1c000    0x1c000    com.apple.kec.corecrypto (12.0) CD18ACA5-5618-3CCF-9232-A8592E5CF16E <7 6 5 4 3 1>
    9    0 0xfffffe00071e8140 0x4000     0x4000     com.apple.kec.AppleEncryptedArchive (1) 1E05C02E-CFFC-38DA-8089-8D8A8E2D8A8E <8 6 5 4>
   10    0 0xfffffe00076a1c50 0x4000     0x4000     com.apple.kec.InvalidateHmac (1) 56F0C2F7-4A49-30F5-9BF7-C230F3FC4F97 <7 6 5 4 3>
   11    6 0xfffffe00076a94f0 0x8000     0x8000     com.apple.kec.Libm (1) D62F6C4A-E2DE-3002-BCDE-6E0957670300 <4>
   12    0 0xfffffe00077abda0 0x4000     0x4000     com.apple.kec.pthread (1) 67CE6458-650B-3522-8E0B-FB6006518C57 <7 6 5 4 3 1>
   13    0 0xfffffe0007019b20 0x4000     0x4000     com.apple.nke.applicationfirewall (402) A121759F-0AE2-3AF2-B359-B1AD6BBE1D97 <7 6 5 4 3 1>
   14   29 0xfffffe00075d7730 0x4000     0x4000     com.apple.iokit.IOReportFamily (47) D58BD9DF-0E66-3130-810B-E748087748BF <6 5 4 3>
   15    1 0xfffffe00075e2240 0x4000     0x4000     com.apple.iokit.IOSlowAdaptiveClockingFamily (1.0.0) C6FB5762-3DE8-3E30-B1BF-E395D8C27B50 <7 6 5 4 3 1>
   16   17 0xfffffe00075e2920 0x4000     0x4000     com.apple.iokit.IOStorageFamily (2.1) FD16B7B3-1FEB-3405-8DAD-31F9B8BAF3F3 <7 6 5 4 3 1>
   17   99 0xfffffe0007043fe0 0x10000    0x10000    com.apple.driver.AppleARMPlatform (1.0.2) 911D503A-285D-36C8-992A-139A9297AA6E <16 15 14 7 6 5 4 3 1>
   18    5 0xfffffe00075660f0 0x4000     0x4000     com.apple.iokit.IOCryptoAcceleratorFamily (1.0.1) DE178630-EA32-3025-A5AA-C90AB2E34DD0 <7 5 4 3 1>
   19    2 0xfffffe0007430300 0x4000     0x4000     com.apple.kext.CoreTrust (1) 3ED8F5F2-CF06-3912-B7C8-0FF9CA75E754 <8 6 4>
   20    2 0xfffffe00072533b0 0xc000     0xc000     com.apple.security.AppleImage4 (4.2.0) F46652D6-4C5A-3160-A3B7-03272FB2ACED <19 18 17 8 7 6 5 4 3 2 1>
   21   14 0xfffffe00072b7af0 0x1c000    0x1c000    com.apple.driver.AppleMobileFileIntegrity (1.0.5) 952AEE17-15F1-32A6-B348-41D54E70962F <20 19 8 7 6 5 4 3 2 1>
   22    2 0xfffffe00072a68a0 0x4000     0x4000     com.apple.kext.AppleMatch (1.0.0d1) 24F2DF59-942C-3E46-90A0-322BDC70F1A1 <4 1>
   23    2 0xfffffe00076bd710 0x20000    0x20000    com.apple.security.sandbox (300.0) 8987E582-DF05-3C28-8170-41317AA334D2 <22 21 16 7 6 5 4 3 2 1>
   24    2 0xfffffe00076b2b50 0x4000     0x4000     com.apple.security.quarantine (4) 775D2050-0C07-31F7-AA34-F20848BDC6FF <23 22 7 6 5 4 3 2 1>
   25    0 0xfffffe00073560d0 0x4000     0x4000     com.apple.AppleSystemPolicy (2.0.0) 5B9D7932-E259-3B5D-ACE2-431139C3F7A6 <24 23 21 7 6 5 4 3 2 1>
   26    5 0xfffffe0007566c60 0x4000     0x4000     com.apple.driver.IODARTFamily (1) 63CB9EC5-CA95-3F3E-9C8C-59792CE63ED6 <17 6 5 4 3 1>
   27   16 0xfffffe00075e5010 0x8000     0x8000     com.apple.iokit.IOSurface (302.14) B51F2E2D-D85F-38BD-9250-62A605B5B0F1 <7 6 5 4 3 1>
   28    4 0xfffffe000726c010 0x30000    0x30000    com.apple.driver.AppleM2ScalerCSCDriver (265.0.0) 451138C3-8A59-3FFE-BA20-BCB94AF0E45C <27 26 17 7 6 5 4 3 1>
   29    3 0xfffffe00070eaf70 0x4000     0x4000     com.apple.driver.AppleAudioClockLibs (140.1) 0D4FF7BA-AEE8-3660-9842-419CB085F0BB <7 5 4 3 1>
   30    6 0xfffffe00072517c0 0x4000     0x4000     com.apple.driver.AppleIISController (140.1) A1B855A8-9C38-3978-AC64-D9CB40875288 <17 7 6 5 4 3 1>
   31    7 0xfffffe000753fe20 0x4000     0x4000     com.apple.iokit.IOAudio2Family (1.0) 4042F1F5-A5C3-34F0-A49E-3E000191AD31 <5 4 3 1>
   32    2 0xfffffe0007565720 0x4000     0x4000     com.apple.iokit.IOCECFamily (1) A2737328-CF7C-3BF4-87A9-D7982B782AC0 <4 3>
   33    5 0xfffffe000758d960 0x8000     0x8000     com.apple.iokit.IOHDCPFamily (1.0.0) 0028295F-9DD6-3F65-9605-03F275E88AAB <5 4 3>
   34   12 0xfffffe0007504c50 0x24000    0x24000    com.apple.iokit.IOAVFamily (1.0.0) A71582E1-E1F4-352C-A0AE-3A90E5B17E5E <33 32 31 30 29 17 11 7 6 5 4 3 1>
[... cut ...]
  240    2 0xfffffe00077bd700 0x4000     0x4000     com.apple.kext.triggers (1.0) 7C659EAA-EEC9-3109-8DB4-E93FB895DEDD <7 6 5 4 3 1>
  241    0 0xfffffe000776b890 0x4000     0x4000     com.apple.filesystems.autofs (3.0) 2E921F78-9428-3B77-9A6C-1680031843D0 <240 7 6 5 4 3 2 1>
  242    0 0xfffffe00076a06f0 0x4000     0x4000     com.apple.driver.driverkit.serial (6.0.0) 1A0E8F48-1D9A-3506-8F50-70FFB330DD22 <44 5 4 3 1>
  244    0 0xfffffe00077acb50 0x10000    0x10000    com.apple.filesystems.smbfs (4.0) C35FF09E-CA77-3337-9F5B-39538E87FF32 <240 8 7 6 5 4 3 1>```

monkeymademe, 2022-08-18:
I have been having similar issues with macOS Ventura 13.0 beta (22A5295i) with an Intel Core i5... The Imager worked fine before the update now either I get corrupted SD card warnings or the imager finishes writing to disk but then when I use the SD card nothing has changed, the card still has its old contents.

On a side note I know my SD cards are not corrupted cause I use DD to install an image and it works out fine

maxnet, 2022-08-18:
>I have been having similar issues with macOS Ventura 13.0 beta (22A5295i) with an Intel Core i5...

Have you also used Imager with a non-beta osx version earlier, and did that work properly?

Another user also reported problems that differed by which beta version he used: https://github.com/raspberrypi/rpi-imager/issues/454

Have not tested beta versions, as I have not enough Apple devices to let them run different osx versions.

maxnet, 2022-09-06:
>I have been having similar issues with macOS Ventura 13.0 beta (22A5295i) with an Intel Core i5...

Unable to reproduce with 22A5331f on an Intel Core i5 Mac Mini (with a simple ugreen usb-c card reader and Samsung EVO+ micro SD-card).
Writes image and advanced settings fine.

[image]

maxnet, 2022-11-24:
>With a HIKVISION MD202 external SSD enclosure
>[...]
>So at least in my case it's not due to a weird reader/enclosure that generally doesn't work with the imager, but something M2/Rosetta2-specific.

I do consider NVMe SSD enclosures to be "non-standard" enclosures.
There are more things that can go wrong signal wise at USB 3.1 10 gbps speeds, than at slower USB rates.
And it is quite possible some computers are more susceptible to that than others.
(And the same goes for the all-in-one USB-C hubs, that also feed the video signal, and 100w power supply through the same cable, that some others users seems to be using...)

So, did you also try a normal SD card reader?

And does the following beta works any better?
[Raspberry Pi Imager 1.7.4.dmg](https://www.maxnet.eu/tmp/174-fat/Raspberry%20Pi%20Imager%201.7.4.dmg)

(It only changes the handling of applying "advanced settings" and shouldn't make any difference if you are not using that, but never hurts to try).

miklosakos, 2022-12-25:
1.7.4 still doesn't work on my M1 Mac running macOS 13.1.
```
miklos_akos@pocketable-nuclear-power-plant MacOS % ./rpi-imager
OSX most preferred language: "en_US"
qt.qpa.fonts: Populating font family aliases took 60 ms. Replace uses of missing font family "Roboto" with one that exists to avoid this cost.
qrc:/main.qml:281:21: QML Rectangle: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
Unmounting: "/dev/disk4"
RECEIVED SIZE: 2
SCMRIGHTS
fd received: 53
Zeroing out first and last MB of drive
Done zeroing out start and end of drive. Took 0 seconds
Image URL: "https://github.com/raspberrypi/rpi-eeprom/releases/download/v2022.12.07-138a1/rpi-boot-eeprom-recovery-2022-12-07-vl805-000138a1-network.zip"
Received header: "HTTP/2 302"
Received header: "server: GitHub.com"
Received header: "date: Sun, 25 Dec 2022 14:40:54 GMT"
Received header: "content-type: text/html; charset=utf-8"
Received header: "vary: X-PJAX, X-PJAX-Container, Turbo-Visit, Turbo-Frame, Accept-Encoding, Accept, X-Requested-With"
Received header: "location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/205523096/17c35c86-1e66-4d98-826e-8b0a43011396?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20221225%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20221225T144054Z&X-Amz-Expires=300&X-Amz-Signature=fc143933cd1a8ace38f9d7faf8c6246eba5e583447a771dbaf2857ae39d0cf62&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=205523096&response-content-disposition=attachment%3B%20filename%3Drpi-boot-eeprom-recovery-2022-12-07-vl805-000138a1-network.zip&response-content-type=application%2Foctet-stream"
Received header: "cache-control: no-cache"
Received header: "strict-transport-security: max-age=31536000; includeSubdomains; preload"
Received header: "x-frame-options: deny"
Received header: "x-content-type-options: nosniff"
Received header: "x-xss-protection: 0"
Received header: "referrer-policy: no-referrer-when-downgrade"
Received header: "content-security-policy: default-src 'none'; base-uri 'self'; block-all-mixed-content; child-src github.com/assets-cdn/worker/ gist.github.com/assets-cdn/worker/; connect-src 'self' uploads.github.com objects-origin.githubusercontent.com www.githubstatus.com collector.github.com raw.githubusercontent.com api.github.com github-cloud.s3.amazonaws.com github-production-repository-file-5c1aeb.s3.amazonaws.com github-production-upload-manifest-file-7fdce7.s3.amazonaws.com github-production-user-asset-6210df.s3.amazonaws.com cdn.optimizely.com logx.optimizely.com/v1/events *.actions.githubusercontent.com wss://*.actions.githubusercontent.com online.visualstudio.com/api/v1/locations github-production-repository-image-32fea6.s3.amazonaws.com github-production-release-asset-2e65be.s3.amazonaws.com insights.github.com wss://alive.github.com; font-src github.githubassets.com; form-action 'self' github.com gist.github.com objects-origin.githubusercontent.com; frame-ancestors 'none'; frame-src viewscreen.githubusercontent.com notebooks.githubusercontent.com; img-src 'self' data: github.githubassets.com media.githubusercontent.com camo.githubusercontent.com identicons.github.com avatars.githubusercontent.com github-cloud.s3.amazonaws.com objects.githubusercontent.com objects-origin.githubusercontent.com secured-user-images.githubusercontent.com/ opengraph.githubassets.com github-production-user-asset-6210df.s3.amazonaws.com customer-stories-feed.github.com spotlights-feed.github.com *.githubusercontent.com; manifest-src 'self'; media-src github.com user-images.githubusercontent.com/ secured-user-images.githubusercontent.com/; script-src github.githubassets.com; style-src 'unsafe-inline' github.githubassets.com; worker-src github.com/assets-cdn/worker/ gist.github.com/assets-cdn/worker/"
Received header: "content-length: 0"
Received header: "x-github-request-id: F73D:F972:83B0404:88BE327:63A8612C"
Received header: ""
Received header: "HTTP/2 200"
Received header: "content-type: application/octet-stream"
Received header: "content-md5: gp57ff9oJLc1i5a+D8RctA=="
Received header: "last-modified: Wed, 07 Dec 2022 17:10:43 GMT"
Received header: "etag: \"0x8DAD875F9B70058\""
Received header: "server: Windows-Azure-Blob/1.0 Microsoft-HTTPAPI/2.0"
Received header: "x-ms-request-id: 79379996-701e-004c-7d6e-18b255000000"
Received header: "x-ms-version: 2020-04-08"
Received header: "x-ms-creation-time: Wed, 07 Dec 2022 17:10:43 GMT"
Received header: "x-ms-lease-status: unlocked"
Received header: "x-ms-lease-state: available"
Received header: "x-ms-blob-type: BlockBlob"
Received header: "content-disposition: attachment; filename=rpi-boot-eeprom-recovery-2022-12-07-vl805-000138a1-network.zip"
Received header: "x-ms-server-encrypted: true"
Received header: "fastly-restarts: 1"
Received header: "accept-ranges: bytes"
Received header: "date: Sun, 25 Dec 2022 14:41:49 GMT"
Received header: "via: 1.1 varnish"
Received header: "age: 54"
Received header: "x-served-by: cache-hhn-etou8220075-HHN"
Received header: "x-cache: HIT"
Received header: "x-cache-hits: 1"
Received header: "x-timer: S1671979309.925914,VS0,VE96"
Received header: "content-length: 743305"
Received header: ""
Write error: "Cannot allocate memory" while writing len: 1048576```
Using the ancient 1.6.2-based Universal build however still works as expected.

miklosakos, 2022-12-25:
I did some testing and turns out that it burns media fine as long as it's a local file. How? I don't know but at that point I can just use dd.

maxnet, 2022-12-25:
>I did some testing and turns out that it burns media fine as long as it's a local file. How?

>Write error: "Cannot allocate memory" while writing len: 1048576

"Cannot allocate memory" is likely to refer to a fixed size memory pool running out somewhere in the Mac OS X kernel.
As applications themselves are less likely to run out of application memory, because system can simply create more virtual memory by swap file for applications.

It could be that the special driver for your Macbook Pro SD card reader is allocating kernel memory out of the same pool as the kernel network code, and it only runs out of memory if those are used concurrently.
But it can also be that it a race condition bug that only occurs if events occur at very specific times.
That's why it is hard to tell if the universal build is really the solution.
Could be that it just happens to work because timinig is different. But that is not a real fix for the real problem, which I suspect is in the OS X kernel.
But hard to tell. Never managed to reproduce the issue.

miklosakos, 2022-12-25:
I don't know what's going wrong but I can 100% say it's rpi-imager being broken. I just checked out Balena Etcher (running through Rosetta as well!) and that was able to not just burn an SD Card properly but not have issues downloading an image either.
I just checked, macOS 10.13 is ancient, to not alienate Apple Silicon users my recommendation is to move to Qt6 and building universal binaries since 1.6.3 built as a Universal binary worked fine on my machine, while the available Intel only builds don't work.
And as we discussed earlier RPi Imager failed to burn the image onto multiple devices which suggests it's not a machine and OS specific problem.

maxnet, 2022-12-25:
>I just checked, macOS 10.13 is ancient

Still 4% market share last month: https://gs.statcounter.com/macos-version-market-share/desktop/worldwide

Not sure how that compares to Macbook **Pro** M1 users.
Don't think the issues affects users of other M1 and M2 models.

miklosakos, 2022-12-25:
>  Don't think the issues affects users of other M1 and M2 models.
Here's #398, #513 just to name a few. All of them with a very similar issue on the surface like mine or worse.
Again, the "don't think the issues affects users of other M1 and M2 models" answer doesn't really seem useful, more like hostile.
I would help debug the software if I had any way of doing so but even compiling RPI Imager is going to be a work in itself since Qt5 can't be compiled on macOS 13 and Qt6 breaks the project.

maxnet, 2022-12-25:
>>Don't think the issues affects users of other M1 and M2 models.

>Here's https://github.com/raspberrypi/rpi-imager/issues/398, https://github.com/raspberrypi/rpi-imager/issues/513 just to name a few.

I see multiple different issues, with different symptoms there.
Sislanger has a Mac Mini, yes. However the report that he gets a **READ error** when using a **CUSTOM image**, is quite different from the problem you are seeing.
And to me sounds like a known issue Imager had with Images that had an uncompressed file size that was not dividable by 512 bytes.
That's why I asked if it concerned Kali (which used to produce those invalid images for a while).

glyph has Macbook Pro, like you. (as he mentions 14" and M1 Max)

Not sure what model lantren has.
That the problem doesn't occur when he/she starts the program from command-line, while it does have it when starting it with the icon, does make me think the issue is very timing specific.
That means anything that changes program flow by a split second may cause this issue to happen or not happen.

>I would help debug the software if I had any way of doing so but even compiling RPI Imager is going to be a work in 
>itself since Qt5 can't be compiled on macOS 13 and Qt6 breaks the project.

There do is a Qt 5 version that supports universal builds expected in April.
The Qt folks released one for commercial license customers in April last year, and they tend to make those releases available as open source a year after release.

jel111, 2023-03-01:
I have the 14" M1. I have the same issue. Tried 4 different cards different readers and different pi's. It fails to write. Machine has eaten a couple cards but if I clean them on a 2016 MBP it works. It's definitely the M1 Mac.

maxnet, 2023-03-03:
>I have the 14" M1. I have the same issue. Tried 4 different cards different readers and different pi's. It fails to write.

With the exact same error "error writing image" as the TS had, or a different one?
Have you tried rebooting your Macbook between using the different card readers?
(Just restarting the application may not be enough, if something exhausted a kernel memory buffer. Which may be the case given the "unable to allocate memory" error miklosakos reported earlier).

>Machine has eaten a couple cards but if I clean them on a 2016 MBP it works.

Do you mean writing the image on an Intel Mac does work?
Or did you mean by "clean", that if you "erase" the card first on your Intel Mac, it then writes succesfully on your M1 Mac?

miklosakos, 2023-08-15:
The issue still persists, release an Apple Silicon specific binary to test. Rebooted, tried multiple different ways of burning media, all fail with the same unable to allocate memory spewed into the terminal.
This is a deal breaker and none of you care enough to fix it.

maxnet, 2023-08-16:
>release an Apple Silicon specific binary to test.

Can find those under the "releases" tab, labeled "universal"
https://github.com/raspberrypi/rpi-imager/releases

Not sure if that solves your issue though.
Never managed to reproduce it.
(Mostly recently tested on a macbook pro that I borrowed from a customer earlier this month. Imager ran fine under Rosetta)

wassup-, 2023-10-11:
> > release an Apple Silicon specific binary to test.
> 
> Can find those under the "releases" tab, labeled "universal" https://github.com/raspberrypi/rpi-imager/releases
> 
> Not sure if that solves your issue though. Never managed to reproduce it. (Mostly recently tested on a macbook pro that I borrowed from a customer earlier this month. Intel Imager ran fine under Rosetta)

The universal build works for me on a Macbook Pro M2.

franknb, 2023-11-19:
> Had this issue while using external card reader (ugreen something), solved by using samsung microsd to sd adapter directly.

I can confirm this. I'm using M1 Max MacBook and Sandisk TF card with UGREEN card reader and it keeps erroring out.
After switching to use SD adapter and inserting onto MacBook directly it works for me.

tdewey-rpi, 2024-09-18:
Closing as stale - wide reports of Apple Silicon writes working OK (including on my own development machine), with additional reports of using third party hardware causing issues - however these are _not_ problems with `rpi-imager`.

closed 2024-09-18 by tdewey-rpi