raspberrypi/rpi-imager #467: Compress images

symbios24, 2022-08-08:
Hello

Can you add an option to compress selected/custom images to squashfs format(berryboot) ??

And or to create a multiboot sd card hdd installation (installs berryboot) with the given selected image and converts the image to squashfs after berryboot is installed.

Thank you

maxnet, 2022-08-08:
>Can you add an option to compress selected/custom images to squashfs format(berryboot) ??

Negative.
Cannot read ext4 file systems on Windows/OSX without a lot of effort.

symbios24, 2022-08-08:
No problem
At least when you have time can you update berryboot kernel/firmware to work correct/better in newer raspbian bullseye operating systems ??

closed 2022-08-08 by maxnet

symbios24, 2022-08-08:
Or create a berryboot automatic installer sh script
So people wont bother you anymore and do it themselves.i know you have more important things than this
Thanks in advance