raspberrypi/rpi-imager #468: IMAGER_REPO_URL is not working with HTTP boot

Mocem1, 2022-08-11:
Hi,

Today I was trying to load my own OS list with the HTTP boot function. The HTTP boot is working fine with the default builtin OS list. My own JSON OS list is only working with rpi-imager at an Ubuntu desktop. 

According to this [topic](https://forums.raspberrypi.com/viewtopic.php?t=329324&start=125) at the official forum I'm not the first person who ran into this issue. The problem has something to do with /sys/bus/nvmem/devices/rmem0/nvmem. 

Best regards,
Marco

maxnet, 2022-08-11:
>According to this [topic](https://forums.raspberrypi.com/viewtopic.php?t=329324&start=125) at the official forum I'm not the first 
>person who ran into this issue. The problem has something to do with /sys/bus/nvmem/devices/rmem0/nvmem.

Has been fixed quite a while ago.
And it does detect it if I set IMAGER_REPO_URL in EEPROM, as you can see from the "using custom repository" line that is displayed in that case:

[image]

What exact EEPROM setting do you have? (run: `rpi-eeprom-config` to see)

lurch, 2022-08-11:
Just a random guess at the cause of your problem, but I remember [reading](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#HTTP_HOST) that the `HTTP_HOST`, `HTTP_PORT` and `HTTP_PATH` settings only support `http://` and not `https://` URLs. Is the same true for the `IMAGER_REPO_URL` setting @maxnet ?

> The problem has something to do with /sys/bus/nvmem/devices/rmem0/nvmem

I suspect that's a red herring?

maxnet, 2022-08-11:
>Is the same true for the IMAGER_REPO_URL setting @maxnet ?

No.
https is fine, provided a proper TLS certificate is used.
Unlike the boot firmware we are not size constrained, so do verify it is a proper certificate issued by a CA and not self-signed stuff.

In any case, even if Imager fails to download or parse the json, it should still show "using custom repository" on the screen (like in my testtesttest.com screesnhot)
Which indicates at least the EEPROM setting itself is detected properly.

timg236, 2022-08-11:
HTTP_HOST, HTTP_PORT and HTTP_PATH don't; support HTTPS (except for our image) because there's no space in the EEPROM for custom cert-stores. IMAGE_REPO_URL is just a string and is ignored by the bootloader.
I'm pretty sure the image supports HTTPS so long as you aren't using a weird HTTPS cert

Mocem1, 2022-08-11:
I have found that the URL is not being saved correctly in the EEPROM. Don't know why yet. Tomorrow I'm going to look into it again. Thanks for a help so far. Good to know this have been addressed already.

closed 2022-08-11 by Mocem1

timdows, 2023-09-11:
Hi @Mocem1 , how did you solve this in the end?

I have set it up to download my json OS list from an ip address (e.g. http://192.168.8.100/oslist.json), but this is not working as execpted.
The logs of my webserver also indicate that there is no get request from the PI itself that it tries to download this list.

Mocem1, 2023-09-11:
Hi @timdows, 

I don't recall it exactly but I think it was by wiping the EEPROM to factory defaults and then apply the settings.

Marco

maxnet, 2023-09-11:
>I have set it up to download my json OS list from an ip address (e.g. http://192.168.8.100/oslist.json), but this is not working as execpted.

If the EEPROM setting was set and detected correctly it should show a text message that it is using a custom repository in the Imager screen.
Is that the case?

timdows, 2023-09-11:
I did get that it wanted to get the file from an external server
But I've used an not that usual port (i.e. 6666)

[image]

Might that be the cause to this issue?

I'll test it tonight to find out myself if there is any change if I go to port 80

maxnet, 2023-09-11:
One setting at a time...

Does the Pi have Internet access?
If so, try setting just IMAGER_REPO_URL first...
And not HTTP_HOST/PORT

To let it do download the Imager software from the Pi website, but use your repository for the json....

maxnet, 2023-09-11:
You are still doing too many changes at once.

Does it work if you maintain timdows.com instead of IP, but do but set http_server ?

Does your dhcp server logs shows it handed out an IP after Imager booted?
(Can be a different one than the firmware used during netinstall)

timdows, 2023-09-13:
> Does it work if you maintain timdows.com instead of IP, but do but set http_server ?

No this didn't work unfortunately

> Does your dhcp server logs shows it handed out an IP after Imager booted? (Can be a different one than the firmware used during netinstall)

I used dnsmasq with these settings
[image]

> Does the OS list not populate after waiting a minute either? (if it takes long time, and you are using a managed Ethernet switch, it could be that have STP enabled on it. If you want to use that, do make sure it is set to portfast/rstp flavor).

After changin the DHCP server for an external one (i.e. my home wifi router) it all worked fine.
I will later this week see if I can reproduce this issue by having the same DHCP server as the IMAGE_REPO_URL