raspberrypi/rpi-imager #469: Error running authopen to gain access to disk device '/dev/rdisk2' on macOS

seguri, 2022-08-11:
I get this error on macOS 11.6.7, imager 1.7.2
Imager has full disk access.
I tried two different card readers. Unfortunately I have no other uSD cards to try.

[image]

[image]

maxnet, 2022-08-11:
Strange.
If you scroll down the list (so you can see what other entries there are below Google chrome), is there perhaps a second "Raspberry Pi Imager" entry listed there?

As it normally shows up as just "Raspberry Pi Imager" in the list, without the ".app" your entry has:

[image]

seguri, 2022-08-11:
Nope, no second entry in any list...

maxnet, 2022-08-11:
Did it work at any point before? E.g. before an OS X update?
Any (corporate) security software installed?

Can also try removing the existing entries with "-".
If there are no longer entries at all for an app (not any entries for the program with unchecked boxes either), OS X should prompt for permission again when you use Imager.

seguri, 2022-08-11:
A previous version of the imager was working, unfortunately I don't know which one.
I have corporate security software installed but that hasn't changed, AFAIK.
I removed all entries, the imager asked access for detachable volumes, I granted it, and it failed as before.

maxnet, 2022-08-11:
If you start a terminal and type: "/usr/libexec/authopen /dev/rdisk2 |hexdump -n 100" it also denies access instead of giving a hexdump?

"diskutil list physical" does show a normal partition layout and returns to shell, and it does not hang while executing?

seguri, 2022-08-11:
`authopen` requires sudo permissions, and hangs.
`diskutil` shows the following, and hangs.
```
/dev/disk2 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:     FDisk_partition_scheme                        *8.1 GB     disk2
   1:             Windows_FAT_32 boot                    268.4 MB   disk2s1
   2:                      Linux                         7.8 GB     disk2s2
```
tomorrow I'll get a new uSD delivered, and I'll repeat all the tests.

maxnet, 2022-08-11:
>not sure if diskutil in both tests is really hanging, but what could it be waiting?

No idea, because it always completed instantly for me.
But there did was another user that reported it hanging for an hour, that is why I was curious if it may be related to the issue you are seeing.

https://github.com/raspberrypi/rpi-imager/issues/270#issuecomment-999326993

closed 2022-08-12 by seguri