raspberrypi/rpi-imager #470: firstrun.sh faild with error message "failed to get default target"

andreasbehnke, 2022-08-12:
Hi,

I try to use the advanced settings and set
- SSH Public Key
- Default user and Password
- WLAN
- Keyboard Layout

I have no success, the raspberry pi will not be configured at first boot.
I can see the message "failed to get default target, could not set address" when firstrun.sh is executed.

I Have tried the following:
- Use different SD Cards
- Only set SSH Public Key but no other settings

After flashing the SD Card, I can see the firstrun.sh Script in the boot Partition and the correct content of cmdline.txt.
After the (failing!) first boot, the SD Card does not contain the firstboot.sh script any more.

So what I can say is:

- The script "firstboot.sh" is being created
- For some reason, the script contains a bug
- The bug does not prevent some part of the script being executed and the script to be deleted
- The Raspberry pi is not correctly configured after that first run

I am wondering about this: The bash reports the error in a line number > 100 (do not know the exact number), but the script is not that long.

maxnet, 2022-08-12:
>After the (failing!) first boot, the SD Card does not contain the firstboot.sh script any more.

If firstboot.sh is no longer there, that means it did run the entire script, as removing it is done on the last line.
So the message you saw was not fatal in any way.

So suggest you double check if really zero of the settings you set got applied.

maxnet, 2022-08-12:
>I am wondering about this: The bash reports the error in a line number > 100 (do not know the exact number), but the script is 
>not that long.

Got a photo?

If it reboots to fast, just open firstrun.sh in a text editor, and manually put a "sleep 30" line at end of the file to delay things.
(or at other random places in the script, if other output fills screen so you can no longer see the message)

andreasbehnke, 2022-08-12:
I rechecked all changes:

- System boots into "autologin". I set a user name (admin) and a password, so would not expect booting into autologin.
- My public SSH key has been set
- WIFI has not been configured

To speed up the debugging process, i stored cmdline.txt and firstrun.sh script, hopefully I can simulate the first boot without flashing again. I will place a wait at the end of the file and take a photo of the screen.

andreasbehnke, 2022-08-12:
ok, the re-run with replacing cmdline.txt and firstboot.sh on the boot partition does not work. System hangs with black screen and blinking cursor. I have to reflash and take the photo.

maxnet, 2022-08-12:
>I set a user name (admin) and a password, so would not expect booting into autologin.

It do is normal that autologin stays enabled.

>WIFI has not been configured

Is there really nothing set in /etc/wpa_supplicant/wpa_supplicant.conf , or you are making the assumption because it does not have connection?
It is very easy to make small but fatal mistakes in wifi configuration. E.g. not everybody realizes the SSID is CaSe SenSitIve.

>ok, the re-run with replacing cmdline.txt and firstboot.sh on the boot partition does not work.

May need to remove the init=script-that-resizes-partitions.sh part from cmdline.txt on subsequent runs.

andreasbehnke, 2022-08-12:
> > I set a user name (admin) and a password, so would not expect booting into autologin.
> 
> It do is normal that autologin stays enabled.
ok, did not expect that because it es quite unsecure

> Is there really nothing set in /etc/wpa_supplicant/wpa_supplicant.conf , or you are making the assumption because it does not have connection? It is very easy to make small but fatal mistakes in wifi configuration. E.g. some do check if their password is correct 10 times, but overlook that they made a mistake in the SSID which is CaSe SenSitIve, "mynetwork" and "MYNETWORK" are not the same.
wpa_supplicant.conf is in place but not working. So I have to investigate into that. I have a very simple SSID and I just used the String provided by the rpi-imager (the same I use for my notebook). Password is a clean copy-past from my password manager.
> 
> > ok, the re-run with replacing cmdline.txt and firstboot.sh on the boot partition does not work.
> 
> May need to remove the init=script-that-resizes-partitions.sh part from cmdline.txt on subsequent runs.
I removed that, but did not help.

andreasbehnke, 2022-08-12:
@maxnet thank you very much for your help!
It turns out, that I'm using WPA3 for the secure network and raspberry pi 3 b does not support this...

I think I was totally missleaded by the error message.

Perhaps it is a good idea to disable autologin if user supplied username and password in the configuration? This was quite 
irritating to me.

closed 2022-08-12 by andreasbehnke

lurch, 2022-08-12:
ping @spl237 for his thoughts on disabling autologin...

XECDesign, 2022-08-12:
Future images should have autologin disabled on lite images which are expected to run headless more often, but will remain on in the desktop images.