raspberrypi/rpi-imager #471: Build Cannot Be Done

TEnLOcODE, 2022-08-13:
I'm trying to build the software but I cannot. I installed anything needed but in the build menu there isn't a build all. it is cannot find cmake or something like that.. I really need help.
Thanks.

maxnet, 2022-08-13:
You will have to be a bit more specific about under which operating system you are trying to build.

TEnLOcODE, 2022-08-13:
Im on windows and I am trying to build it to windows.

maxnet, 2022-08-13:
As stated in the README.md, on Windows you need to install Qt with CMake and Mingw.

>Get dependencies
>Get the Qt online installer from: https://www.qt.io/download-open-source During installation, choose a Qt 5.x with Mingw32 
>32-bit toolchain and CMake.

That really should not be more complicated than checking the box that says "CMake" in the installer, like this:

[image]

If for some reason, it still claims it cannot find CMake, you may want to ask around in a Qt forum instead.
As in that case, I don't think that will be an issue specific to rpi-imager.

TEnLOcODE, 2022-08-23:
Thanks, Unfortunately, I already installed qt so I'll need to uninstall and re-install the program

maxnet, 2022-08-23:
No, you don't
There's also "maintain Qt" to add/remove components later.

Anyway, closing this, as this is not an Imager issue.

closed 2022-08-23 by maxnet