raspberrypi/rpi-imager #473: Raspberry Pi Imager v 1.7.2: "Error writing file to disk"

dieseltravis, 2022-08-17:
I'm trying to use Raspberry Pi Imager v1.7.2 on Windows 10 Pro, 64-bit 10.0.19044 to install Home Assistant OS Installer for Yellow. After selecting the disk to write to, selecting Write, and then Yes, I get an error message "Error writing file to disk". I have tried with both a Generic 2GB and Sandisk 128GB USB drive (pictured below) and both give me the same error.

[image]
[image]
[image]

maxnet, 2022-08-17:
@agners
Could you compress the "Installer for Home Assistant Yellow Kit" image, so that it is a .img.xz/.img.gz/.zip file?

It is currently not compressed, and Imager has some issues on the Windows platform when streaming uncompressed files from the Internet.
(Imager tries to write the image data instantly in whatever chunk size it gets from the server while it downloads. But Windows only accepts writes that are a multiples of the 512 bytes sector size.
When dealing with compressed images, we have a different workflow that buffer things and this issue does not occur)

agners, 2022-08-17:
@maxnet uh interesting! Sounds like a RPi Imager bug to me, but I can of course create a compressed image :smile: I'll have a look into it tomorrow.

Btw, we just today experienced what looks like https://github.com/raspberrypi/rpi-imager/issues/466. Could that be related to the fact that the image is uncompressed?

maxnet, 2022-08-17:
>Sounds like a RPi Imager bug to me

It certainly is, and should be fixed.
But working around it is quicker than doing both that and convincing people to update to new version.

>Could that be related to the fact that the image is uncompressed?

Quite possible.
Recall Mac OS X does not like writes that are off sector size either.

(Linux however happily accepts writes in any size, so that's why it escaped testing).

maxnet, 2022-08-18:
This build does support streaming uncompressed images from the Internet: [imager-1.7.3.exe](https://www.maxnet.eu/tmp/173-buffer/imager-1.7.3.exe)

lurch, 2022-08-18:
> I can of course create a compressed image 

Given the amount of bandwidth and download-time it would save, I dunno why you would _want_ to stream an uncompressed image from the internet? :upside_down_face:

dieseltravis, 2022-08-18:
@maxnet this works for me, thanks!

agners, 2022-08-31:
> Given the amount of bandwidth and download-time it would save, I dunno why you would _want_ to stream an uncompressed image from the internet? :upside_down_face:

Well, what is faster depends heavily on the Internet speed. xz decompression is actually really not that fast, especially slow CPUs. I haven't done my own benchmark, but I've read 10MB/s on a RPi 4, so 100MBit Internet and faster makes uncompressed image actually the faster option. :sunglasses: 

But the main reason I opted for uncompressed are it's size, the image is very small (64MiB). And the fact that it uses a compressed squashfs inside (to boot the system in memory only, as this is an installer). So compressing the complete image again is somewhat weird.

In any case, I created a new release which also makes the image 1MB aligned, and xz compressed. Tests showed that this works now with MacOS systems.

lurch, 2022-08-31:
> Also, the image is very small (64MiB).

Ahh, fair enough :wink:

closed 2022-09-06 by maxnet