raspberrypi/rpi-imager #474: Underscores are allowed in Raspbian hostnames, but they cause bugs in the actual OS

manyet1k, 2022-08-18:
[image]
In this box, the user can add an underscore '_' to their hostname, but this causes network issues. I believe that _ is not a valid character for a hostname. The program shouldn't let the user save the settings if there is an _, or any other illegal characters, in the textbox.

lurch, 2022-08-18:
Presumably this is a problem that @spl237 has already had to solve in raspi-config / rc_gui ?

spl237, 2022-08-18:
One for whoever maintains the imager tool - Floris?

andrum99, 2022-08-18:
Per the warning in `raspi-config` when you go to change the hostname, underscores in hostnames are not valid. Assuming that is correct, then imager should prevent underscores in hostnames.

maxnet, 2022-08-18:
>Per the warning in raspi-config when you go to change the hostname, underscores in hostnames are not valid. Assuming that is 
>correct, then imager should prevent underscores in hostnames.

Yes, and probably also the other restrictions on DNS host name per https://www.ietf.org/rfc/rfc1035.txt 
Like 63 character limit for a host name component (label).

commit b09fc414 (2022-08-18): Prevent user from entering invalid characters in hostname field

Implemented as "validator" which will cause that if an user
presses a key that is not allowed, it will not register.
Saves having to add an error message that would need to be
translated to different languages.

Closes #474
commit ce0b02b8 (2022-08-18): Prevent user from entering invalid characters in hostname field

Implemented as "validator" which will cause that if an user
presses a key that is not allowed, it will not register.
Saves having to add an error message that would need to be
translated to different languages.

Closes #474
closed 2022-08-18 by maxnet