raspberrypi/rpi-imager #477: Add 'website' parameter to RaspberryMatic entry in os_list_imagingutility_v3.json

jens-maus, 2022-08-23:
In the latest https://downloads.raspberrypi.org/os_list_imagingutility_v3.json which the RaspberryPi Imager uses, the respective entry for the RaspberryMatic project looks like:

```
                        {
                            "name": "RaspberryMatic",
                            "description": "Lightweight Linux OS for running a HomeMatic/homematicIP IoT central.",
                            "icon": "https://raw.githubusercontent.com/jens-maus/RaspberryMatic/master/release/rpi-imager.png",
                            "subitems_url": "https://raw.githubusercontent.com/jens-maus/RaspberryMatic/master/release/rpi-imager.json"
                        },
```

Unfortunately the `website` parameter is missing. So I would like to ask/request to please add it accordingly so that the RaspberryMatic JSON will look like:

```
                        {
                            "name": "RaspberryMatic",
                            "description": "Lightweight Linux OS for running a HomeMatic/homematicIP IoT central.",
                            "icon": "https://raw.githubusercontent.com/jens-maus/RaspberryMatic/master/release/rpi-imager.png",
                            "subitems_url": "https://raw.githubusercontent.com/jens-maus/RaspberryMatic/master/release/rpi-imager.json",
                            "website": "https://raspberrymatic.de/"
                        },
```

This will then result in the small (i) symbol being displayed in the RaspberryPi Imager right beside the RaspberryMatic entry so that users can click on it an get more detailed information about it.

XECDesign, 2022-08-23:
Added

jens-maus, 2022-08-23:
Thanks, works!

closed 2022-08-23 by jens-maus