raspberrypi/rpi-imager #479: Proxy support

chromer030, 2022-09-09:
Please add proxy support for downloading os list and os download , for systems behind firewall or restricted network.
I know we can use environment variables but it sometimes works and sometimes not.
Thanks.

maxnet, 2022-09-09:
It do is supposed to get supposed the proxy settings from the operating systems. At least on operating systems that support setting it there.
(since v1.7 https://github.com/raspberrypi/rpi-imager/commit/e79467b6264d96540ad4786028813d7b3cf660fb )

What operating system are you using?

chromer030, 2022-09-09:
The point of this issue is not finding workaround , as there is some, like VPN, tools like tun2socks , tsocks , proxychain etc.
It's about internal feature and switch for setting the proxy inside the GUI or command line.
If it'not possible or not in the milestone , this issue can be ignored and closed.
Thanks a lot.

maxnet, 2022-09-09:
>Arch Linux , It's sometimes working and sometimes not , i don't know why.
>tried explicit variable : $ HTTPS_PROXY=127.0.0.1:1080 rpi-imager

Try it with https_proxy in lowercase.

Recall that if Qt is compiled with libproxy support (no idea wheter or not that is the case on Arch Linux), it should be able to get proxy system settings on the Linux platform.
Would argue that if Arch Linux does not offer a system-wide proxy setting, that may be an Arch Linux problem, rather than an Imager problem.

omniproc, 2025-08-03:
Since auto-discovery and configuration ultimately will never work on all systems due to missing standards how about just making it possible for a user to configure proxy settings on his own?